openbmc/openbmc #2118: org.open_power.OCC.Control.service: Failed with result 'core-dump' [ Witherspoon]

gkeishin, 2017-08-08:
- [x] Boot Host
- [x] Reset BMC
- [x] Host booted back OK

but see's OCC service core dumped. This is with RR fix https://gerrit.openbmc-project.xyz/#/c/5805/

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-192"
VERSION_ID="v1.99.8-192-g4e9d5e7"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-192"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-baeb50beb784ac5993d2bfb5d06d909971b60506 #1 Tue Aug 8 13:22:17 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/136cf504": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.17-210-gbabb528,buildroot-2017.05-93-g9b91216,skiboot-v5.7,hostboot-f523eaa,linux-4.12.1-openpower1-p8927fdf,petitboot-v1.5.1-pce8abef,machine-xml-5575e7e,occ-ca84830,hostboot-binaries-c24fccd,capp-ucode-9c73e9f,sbe-02021c6-p6a0f00f",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.87"
    },
    "/xyz/openbmc_project/software/d39e43b7": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-192-g4e9d5e7"
    }
}
```
- [x] Journald 
[journald_v1.99.8-192-g4e9d5e7_OCC_coredump.txt](https://github.com/openbmc/openbmc/files/1208675/journald_v1.99.8-192-g4e9d5e7_OCC_coredump.txt)

geissonator, 2017-08-08:
Hmm, we correctly detected the power was on, but did not get a reply from the host that it was running:
```
Aug 03 15:45:01 witherspoon ipmid[1242]: Host control timeout hit!
Aug 03 15:45:01 witherspoon ipmid[1242]: Timer expired
Aug 03 15:45:01 witherspoon phosphor-host-check[1485]: Host control signal values
Aug 03 15:45:01 witherspoon phosphor-host-check[1485]: Host is not running!
```

geissonator, 2017-08-09:
@gkeishin - not clear based on  your data above, but when we reset the bmc, with the host up, the host should stay up the entire time.  So you'll want to look at "uptime" or get an ssh session into the OS to verify it stays up.  Based on some of my own testing today, I think you need https://gerrit.openbmc-project.xyz/#/c/5860/ for that to all work.  We'll track the OCC crash with this issue, which is a separate problem.

geissonator, 2017-08-09:
When I booted, I saw openpower-occ-control core dump.  It seemed like when we got the inventory for the procs, we expected the sysfs dirs to be present (like in occ_device).  That’s no longer the case now that we need to do this secondary hub scanning.  Probably need some sort of solution where we don’t look at any of the occ filesystem until we know this secondary hub scan is done.
I figured I could just add the scan before we booted, but the host died with FSI errors, so seems like it’s required after.  Once booted, I manually did the scan and then started openpower-occ-control.  This worked ok, but we didn’t do the bind because we missed the occ active setting from the host, another thing to think about (probably going to need to persist the occ status)

I put a simple service up in gerrit to get things started with this secondary hub scan - https://gerrit.openbmc-project.xyz/#/c/5862/

Note that my testing included the patch from Eddie in https://gerrit.openbmc-project.xyz/#/c/5098/

vishwabmc, 2017-08-09:
@geissonator I am thinking that occ-control application writing a `1` to the `rescan` file would make sure that the needed `/sys` files are created before occ-control goes ahead accessing those.

geissonator, 2017-08-09:
#2126 opened to track the persistency aspect.

rfrandse, 2017-08-09:
<https://gerrit.openbmc-project.xyz/5897> Call Hub FSI scan service prior to doing OCC bind
Fixes  #2118 org.open_power.OCC.Control.service: Failed with result 'core-dump' [ Witherspoon]

vishwabmc, 2017-08-10:
# **STATUS**

closed 2017-08-11 by williamspatrick