openbmc/openbmc #2121: Host power on process?

sdssly, 2017-08-09:
Could someone point me the detailed information about the host power on process?

My understanding is: 
1.	When press the power button, the button_power will trigger a GPIO interrupt and emit the press/release event
2.	Chassis_control.py will handle the signal and start the obmc-host-start@0.target 

My confusion is: In order to power on the host, we must setPowerState to 1, to change the power_up_outs gpio pins for host power on.

But I did not find who calls the setPowerState during the power on process

gtmills, 2018-05-24:
@geissonator You could explain this better than anyone. Thanks! 
@sdssly This is a good question for the mailing list, as it's not a bug for or feature request. You can sign yourself up at for the list at https://lists.ozlabs.org/listinfo/openbmc

geissonator, 2018-07-09:
https://github.com/openbmc/docs/blob/master/openbmc-systemd.md has some details on the systemd targets we use. On a witherspoon system, here's the service that does it:
```
# cat op-power-start@0.service
[Unit]
Description=Start Power%i
Wants=obmc-power-start@%i.target
Before=obmc-power-start@%i.target
After=obmc-power-start-pre@%i.target
After=obmc-fan-control.target
Wants=mapper-wait@-org-openbmc-control-power%i.service
After=mapper-wait@-org-openbmc-control-power%i.service
Conflicts=obmc-chassis-poweroff@%i.target
ConditionPathExists=!/run/openbmc/chassis@%i-on

[Service]
RemainAfterExit=yes
Type=oneshot
ExecStart=/bin/sh -c "busctl call `mapper get-service /org/openbmc/control/power%i` /org/openbmc/control/power%i org.openbmc.control.Power setPowerState i 1"
SyslogIdentifier=op-power-start

[Install]
WantedBy=obmc-host-start@%i.target
```

gtmills, 2018-07-09:
@geissonator Thanks for the reply!
@sdssly Going to close. If you still have problems please reopen. In the future, the mailing list is best for questions like this. Thanks!

closed 2018-07-09 by gtmills