openbmc/openbmc #2122: Network:- Default gateway is getting modified with network route.

prkatti1, 2017-08-09:
This get reproduced after adding and deleting the ip.

gkeishin, 2017-08-09:
@prkatti1 Can u add the steps to reproduce this one. 

This one is with patch test on https://gerrit.openbmc-project.xyz/#/c/5813/

prkatti1, 2017-08-09:
@gkeishin It is happeining while adding and deleting IP address:

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["xyz.openbmc_project.Network.IP.Protocol.IPv4","10.6.6.6",24,"0.0.0.0"] }' https://xx.xx.xx.xx/xyz/openbmc_project/network/eth0/action/IP

********************************************************************************

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X DELETE https://xx.xx.xx.xx/xyz/openbmc_project/network/eth0/ipv4/fbfc29b

ratagupt, 2017-08-14:
@prkatti1 :- Can you test this with the latest driver now?

ratagupt, 2017-08-16:
@prkatti1 : Can you update your findings with the latest driver?

gkeishin, 2017-08-16:
@prkatti1 Can u check on this one and update.. we need this one going out if clean. Thanks

prkatti1, 2017-08-16:
it is working now, so closing this.

closed 2017-08-16 by prkatti1