openbmc/openbmc #2124: HOST is not powering on after doing witherspoon-redbud.pnor.squashfs.tar upgrade

PavamanSubramaniyam, 2017-08-09:
I have generated the PNOR image witherspoon-redbud.pnor from master op-build git tree.
Then used the generate-squashfs script to generate the witherspoon-redbud.pnor.squashfs.tar.
Executed the following Rest API functionalities to transfer the squashfs to Open BMC machine and also update the same image.

```
curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://<BMCIP>/login
curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" -X POST -T witherspoon-redbud.pnor.squashfs.tar https://<BMCIP>/upload/image
```

Then fetched the id value using the below enumerate Rest API.
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" https://<BMCIP>/xyz/openbmc_project/software/enumerate
```

Then requested for Activation using the following Rest API.
```
curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active"}' https://<BMCIP>/xyz/openbmc_project/software/e066a32f/attr/RequestedActivation
```

After this the machine is not powering on and is going to Quiesced state.
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced

root@witherspoon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-200"
VERSION_ID="v1.99.8-200-g201bd62"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-200"
BUILD_ID="v1.99.8"
```

PavamanSubramaniyam, 2017-08-09:
```
root@powerkvm3-lp1:~# curl -c cjar -b cjar -k -H "Content-Type: application/json" https://<BMCIP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/aea78f12": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-200-g201bd62"
    },
    "/xyz/openbmc_project/software/e066a32f": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "buildroot-2017.05-163-g8035745,skiboot-v5.7-24-g81c0e076374e-pc665224,hostboot-faac61e,linux-4.12.5-openpower1-pafd8d82,petitboot-v1.5.1-pb361a08,machine-xml-d5a82f8-p9257999,occ-6d29425,hostboot-binaries-883d218,capp-ucode-9c73e9f,sbe-02021c6-p757955a",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "open-power-witherspoon-v1.18-38-gdb19752"
    },
    "/xyz/openbmc_project/software/e49bc78e": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.17-210-g7c61ec6,buildroot-2017.05-93-g9b91216,skiboot-v5.7-paa10293,hostboot-f523eaa,linux-4.12.1-openpower1-p1cc6f7f,petitboot-v1.5.1-p279e9f9,machine-xml-5575e7e,occ-ca84830,hostboot-binaries-c24fccd,capp-ucode-9c73e9f,sbe-02021c6-pfad3bc9",
      "Path": "",
      "Priority": 1,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.88"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

geissonator, 2017-08-09:
After first moving to the new squashfs PNOR, you need to reboot your BMC.  Did you do that in this flow?  Also, please provide the journal (or the .tar generated by ffdc -d /tmp/).

PavamanSubramaniyam, 2017-08-09:
[20170809120604603061_Myffdc.zip](https://github.com/openbmc/openbmc/files/1211349/20170809120604603061_Myffdc.zip)

PavamanSubramaniyam, 2017-08-09:
@geissonator I had executed obmcutil poweroff command in BMC shell and then executed reboot command in the BMC shell to reboot the BMC.

After that I again tried to power on the HOST using obmcutil poweron command.

saqibkh, 2017-08-10:
There are a lot of mboxd failures in the log.
Have you tried rebooting the BMC after a PNOR Update?

Can you reboot the BMC and see if the issue still exists?
If it still exists then we will need @dkodihal to take a look.

PavamanSubramaniyam, 2017-08-14:
I have rebooted the BMC and also tried the same steps again with the latest Open BMC and PNOR Firmware images built from master git trees.
Still the HOST is not powering on and the HOST goes to Quiesced state.

saqibkh, 2017-08-14:
@dkodihal can you take a look at this defect?
There are a bunch of mboxd core dumps in the log.

Aug 14 09:16:55 witherspoon daemon.info mboxd[1082]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 14 09:16:55 witherspoon daemon.info mboxd[1082]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 14 09:16:55 witherspoon daemon.info mboxd[1082]: [ 1502702215.539052331] Partition HBB not found
Aug 14 09:16:55 witherspoon daemon.warn systemd[1]: mboxd.service: Main process exited, code=dumped, status=6/ABRT
Aug 14 09:16:55 witherspoon daemon.notice systemd[1]: mboxd.service: Unit entered failed state.
Aug 14 09:16:55 witherspoon daemon.warn systemd[1]: mboxd.service: Failed with result 'core-dump'.
Aug 14 09:16:55 witherspoon daemon.info systemd[1]: mboxd.service: Service hold-off time over, scheduling restart.
Aug 14 09:16:55 witherspoon daemon.info systemd[1]: Stopped Phosphor MBOX Daemon.
Aug 14 09:16:55 witherspoon daemon.info systemd[1]: Started Phosphor MBOX Daemon.
Aug 14 09:16:56 witherspoon kern.err mboxd[1086]: The operation failed internally.

dkodihal, 2017-08-14:
Partition HBB not found - this means that mboxd did not find the virtual pnor layout. My guess is a step might have been missed while updating this box using the new code update process.

dkodihal, 2017-08-14:
@gkeishin can you help Pavaman with the right steps?

dkodihal, 2017-08-14:
Also, @saqibkh or @anoo1 you were working on a story to have the virtual pnor layout created in the way to standby? That will also enable mboxd to see the vpnor layout with a need to reboot.

saqibkh, 2017-08-14:
@dkodihal we have a story ready for merge that creates a virtual pnor layout on the way to HOST poweron.

Regarding this defect, the system has the correct vpnor layout, correct ubi volumes and dbus objects and has been rebooted multiple times.

dkodihal, 2017-08-14:
@saqibkh the log "Partition HBB not found" means the partition file named "HBB" was not found in the partition file search location. So I'm not sure if the vpnor layout is correct.

PavamanSubramaniyam, 2017-08-16:
@dkodihal I am seeing the HBB file in the machine at the mentioned location
```
root@witherspoon:~# ls -l /var/lib/phosphor-software-manager/pnor/ro/
total 44466
-rw-r--r--    1 root     root         32768 Aug  9 11:11 ATTR_PERM
-rw-r--r--    1 root     root         32768 Aug  9 11:11 ATTR_TMP
-rw-r--r--    1 root     root         36864 Aug  9 11:11 BMC_INV
-rw-r--r--    1 root     root      15728640 Aug  9 11:11 BOOTKERNEL
-rw-r--r--    1 root     root        147456 Aug  9 11:11 CAPP
-rw-r--r--    1 root     root        294912 Aug  9 11:11 CVPD
-rw-r--r--    1 root     root        294912 Aug  9 11:11 DJVPD
-rw-r--r--    1 root     root         12288 Aug  9 11:11 FIRDATA
-rw-r--r--    1 root     root         20480 Aug  9 11:11 GUARD
-rw-r--r--    1 root     root       1048576 Aug  9 11:11 HBB
-rw-r--r--    1 root     root         24576 Aug  9 11:11 HBBL
-rw-r--r--    1 root     root       1179648 Aug  9 11:11 HBD
-rw-r--r--    1 root     root        147456 Aug  9 11:11 HBEL
-rw-r--r--    1 root     root      12976128 Aug  9 11:11 HBI
-rw-r--r--    1 root     root       4718592 Aug  9 11:11 HBRT
-rw-r--r--    1 root     root         20480 Aug  9 11:11 HB_VOLATILE
-rw-r--r--    1 root     root       1179648 Aug  9 11:11 HCODE
-rw-r--r--    1 root     root        262144 Aug  9 11:11 IMA_CATALOG
-rw-r--r--    1 root     root        589824 Aug  9 11:11 MVPD
-rw-r--r--    1 root     root        589824 Aug  9 11:11 NVRAM
-rw-r--r--    1 root     root       1179648 Aug  9 11:11 OCC
-rw-r--r--    1 root     root       1048576 Aug  9 11:11 PAYLOAD
-rw-r--r--    1 root     root        131072 Aug  9 11:11 RINGOVD
-rw-r--r--    1 root     root        532480 Aug  9 11:11 SBE
-rw-r--r--    1 root     root        147456 Aug  9 11:11 SECBOOT
-rw-r--r--    1 root     root          4096 Aug  9 11:11 VERSION
-rw-r--r--    1 root     root       3145728 Aug  9 11:11 WOFDATA
-rw-r--r--    1 root     root          4096 Aug  9 11:11 part
-rw-r--r--    1 root     root          2048 Aug  9 11:11 pnor.toc
```

dkodihal, 2017-08-16:
@PavamanSubramaniyam you're seeing it in the ro/ location. @saqibkh and I saw that as well. It was supposed to be in another location. I guess Saqib asked you to try this with the latest code. Did you get a chance to do that?

PavamanSubramaniyam, 2017-08-16:
I have updated the machine to latest master Open BMC Firmware image.
Then executed all the steps for flashing using curl commands as discussed with @gkeishin.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-256"
VERSION_ID="v1.99.8-256-gb25ba23"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-256"
BUILD_ID="v1.99.8"

root@powerkvm3-lp1:~# curl -c cjar b cjar -k -H "Content-Type: application/json" https://9.40.193.227/xyz/openbmc_project/software/enumeraten
{
  "data": {
    "/xyz/openbmc_project/software/91087e8b": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-256-gb25ba23"
    },
    "/xyz/openbmc_project/software/e066a32f": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "buildroot-2017.05-163-g8035745,skiboot-v5.7-24-g81c0e076374e-pc665224,hostboot-faac61e,linux-4.12.5-openpower1-pafd8d82,petitboot-v1.5.1-pb361a08,machine-xml-d5a82f8-p9257999,occ-6d29425,hostboot-binaries-883d218,capp-ucode-9c73e9f,sbe-02021c6-p757955a",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "open-power-witherspoon-v1.18-38-gdb19752"
    },
    "/xyz/openbmc_project/software/e49bc78e": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.17-210-g7c61ec6,buildroot-2017.05-93-g9b91216,skiboot-v5.7-paa10293,hostboot-f523eaa,linux-4.12.1-openpower1-p1cc6f7f,petitboot-v1.5.1-p279e9f9,machine-xml-5575e7e,occ-ca84830,hostboot-binaries-c24fccd,capp-ucode-9c73e9f,sbe-02021c6-pfad3bc9",
      "Path": "",
      "Priority": 1,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.88"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

PavamanSubramaniyam, 2017-08-16:
I have tried to upgrade the latest witherspoon-redbud.pnor.squashfs.tar generated from upstream op-build git tree using the curl Rest API's method.
But still the machine is not able to power on the HOST and the state is HostState.Quiesced.

geissonator, 2017-08-16:
Please provide a "ffdc" dump from the system.

PavamanSubramaniyam, 2017-08-17:
[20170817052311836056_Myffdc.zip](https://github.com/openbmc/openbmc/files/1230210/20170817052311836056_Myffdc.zip)

geissonator, 2017-08-17:
I believe this is the issue we ran into yesterday with Bill when generating a squashfs from op-build.  @anoo1 has a fix up here - https://gerrit.openbmc-project.xyz/#/c/6045/

closed 2017-08-18 by williamspatrick