openbmc/openbmc #2126: Ensure occ sensors are provided on bmc reset with host up

geissonator, 2017-08-09:
If the BMC is reset while the host is up, when the BMC comes back up it needs to scan the second proc fsi hub, and rebind to the occ's (to get the hwmon sensors back up and running).  To do this, it needs to know that the occ's are active.  So we'll need to persist the occ active state, and upon startup of the occ application, do all the required functions automatically if it is active.  

We will need a validity check against the host state target to ensure the host is running (to handle situations where someone a/c cycles the system with the host up, occ active would still be set but host would not be on).

This issue tracks getting all of this function in place.

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6169> Enable OCC Active on BMC reset reload
Fixes  #2126 Ensure occ sensors are provided on bmc reset with host up

vishwabmc, 2017-08-23:
# **STATUS**

- 8/23:
   - Pushed the change. Awaiting review

closed 2017-08-29 by williamspatrick