openbmc/openbmc #2127: Host failed to boot: phosphor-fan-presence-tach@0.service: Main process exited, code=dumped,

gkeishin, 2017-08-09:
`snippet log`
```
Aug 03 15:51:00 witherspoon systemd[1]: Stopped Phosphor Fan Presence Tach Daemon.
Aug 03 15:51:00 witherspoon phosphor-fan-presence-tach[1398]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 03 15:51:00 witherspoon phosphor-fan-presence-tach[1398]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 03 15:51:00 witherspoon kernel[946]: [  493.640000]  slave@00:00: engine[4]: type 6, version 1, addr 1400 size 400
Aug 03 15:51:00 witherspoon systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Aug 03 15:51:00 witherspoon kernel[946]: [  493.640000]  slave@00:00: engine[5]: type 7, version 2, addr 1800 size 400
Aug 03 15:51:00 witherspoon phosphor-fan-presence-tach[1398]: The operation failed internally.
Aug 03 15:51:01 witherspoon phosphor-fan-presence-tach[1409]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 03 15:51:01 witherspoon phosphor-fan-presence-tach[1409]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 03 15:51:00 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Main process exited, code=dumped, status=6/ABRT
Aug 03 15:51:00 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Unit entered failed state.
Aug 03 15:51:00 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Failed with result 'core-dump'.
Aug 03 15:51:00 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Service hold-off time over, scheduling restart
```

```
● phosphor-fan-presence-tach@0.service - Phosphor Fan Presence Tach Daemon
   Loaded: loaded (/lib/systemd/system/obmc-chassis-poweron@0.target.requires/../phosphor-fan-presence-tach@.service; disabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Thu 2017-08-03 15:51:03 UTC; 10min ago
  Process: 1495 ExecStart=/usr/sbin/phosphor-fan-presence-tach (code=dumped, signal=ABRT)
 Main PID: 1495 (code=dumped, signal=ABRT)

Aug 03 15:51:03 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Main process exited, code=dumped, status=6/ABRT
Aug 03 15:51:03 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Unit entered failed state.
Aug 03 15:51:03 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Failed with result 'core-dump'.
Aug 03 15:51:03 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Service hold-off time over, scheduling restart.
Aug 03 15:51:03 witherspoon systemd[1]: Stopped Phosphor Fan Presence Tach Daemon.
Aug 03 15:51:03 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Start request repeated too quickly.
Aug 03 15:51:03 witherspoon systemd[1]: Failed to start Phosphor Fan Presence Tach Daemon.
Aug 03 15:51:04 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Unit entered failed state.
Aug 03 15:51:04 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Failed with result 'core-dump'.
root@witherspoon:~#
```

`Driver info`
```
/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/136cf504": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.17-210-gbabb528,buildroot-2017.05-93-g9b91216,skiboot-v5.7,hostboot-f523eaa,linux-4.12.1-openpower1-p8927fdf,petitboot-v1.5.1-pce8abef,machine-xml-5575e7e,occ-ca84830,hostboot-binaries-c24fccd,capp-ucode-9c73e9f,sbe-02021c6-p6a0f00f",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.87"
    },
    "/xyz/openbmc_project/software/b346ead3": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-204-g169a8a3"
    }
}

root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-baeb50beb784ac5993d2bfb5d06d909971b60506 #1 Wed Aug 9 14:44:15 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald 
[journald_v1.99.8-204-g169a8a3_fan_crashed.txt](https://github.com/openbmc/openbmc/files/1212106/journald_v1.99.8-204-g169a8a3_fan_crashed.txt)

msbarth, 2017-08-09:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-204"
VERSION_ID="v1.99.8-204-g169a8a3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-204"
BUILD_ID="v1.99.8"

geissonator, 2017-08-09:
Looks like we missed a needed kernel config option:
CONFIG_KEYBOARD_GPIO_POLLED=y

gkeishin, 2017-08-09:
Fix https://gerrit.openbmc-project.xyz/#/c/5898/

eddiejames, 2017-08-09:
Looks like my commit message edit with the "Resolves" didn't work. This is fixed and merged: https://github.com/openbmc/openbmc/commit/6f23d594cf43dc703844740cef42e27867c23327

closed 2017-08-09 by geissonator