openbmc/openbmc #2129: ITC:OP910: SSH is not working on Witherspoon system after latest firmware update.

cdeadmin, 2017-08-09:


geissonator, 2017-08-10:
Going to use this issue to track the dropbear related issues we've seen on openbmc.  This specifically tracks where we're able to ping the bmc IP address and use the REST api's, but not able to ssh into it.

This system was running v1.99.8-98-g06734f31-r2-0-g752ced3

It had been working fine, then suddenly ssh stopped working, it seemed to come back for a while, and then stopped working again.  I got onto the system at that last step through the serial connection and started debug.

Here's an attempt at ssh into the system:
```
$ ssh -vvv root@9.5.180.113
OpenSSH_5.8p2, OpenSSL 1.0.0g 18 Jan 2012
debug1: Reading configuration data /gsa/ausgsa/home/a/n/andrewg/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: Applying options for *
debug2: ssh_connect: needpriv 0
debug1: Connecting to 9.5.180.113 [9.5.180.113] port 22.
debug1: connect to address 9.5.180.113 port 22: Connection refused
ssh: connect to host 9.5.180.113 port 22: Connection refused
```

There was no logs in dmesg or the journal on the bmc when these ssh attempts were made.

On the BMC, there were 4 dropbear session opened:
```
ps -ef | grep drop
 3046 root       0:01 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 3062 root       0:00 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 3078 root       0:00 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 3083 root       0:00 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -c /usr/bin/obmc-console-client -B
```

What ended up fixing everything was when I killed process 3046.  After that, we could open up as many ssh sessions into the system as we liked.

The person on the system prior, who hit the initial issues stated that the trigger for him was when he connected an ssh session to another machine via the host of the system that stopped responding.  This could imply a MAC address duplication or bug in the broadcom NC-SI firmware potentially.

geissonator, 2017-08-11:
# **STATUS**
8/11 - Got some details from Ben on his recreate, will try it out today (ssh via bmc port 2200 into host to a different system)

rfrandse, 2017-09-28:
CQ was rejected as unreproducible

closed 2017-09-28 by rfrandse