openbmc/openbmc #2132: Host is rebooting after powering on with latest master v1.99.8-208-g4ee6757

rahulmah, 2017-08-10:
Host is rebooting after powering on with latest master v1.99.8-208-g4ee6757. Later it is going into quiesce state.

```
curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/state/enumerate
{
  "data": {
    "/xyz/openbmc_project/state/bmc0": {
      "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
      "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
      "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
      "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
      "AttemptsLeft": 3,
      "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
      "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Quiesced",
      "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
      "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Reboot"
```

Fw info.
```
curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/2b6dcac8": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.18-35-g04a11fe,buildroot-2017.05-163-g8035745,skiboot-07937c7,hostboot-faac61e,linux-4.12.3-openpower1-p593bf1b,petitboot-v1.5.1-p1b1d3ef,machine-xml-5575e7e-p28bf2dc,occ-d650501,hostboot-binaries-883d218,capp-ucode-9c73e9f,sbe-02021c6-pea3c159",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.11"
    },
    "/xyz/openbmc_project/software/3c88ccca": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-208-g4ee6757"
    }
  },
  "message": "200 OK",
  "status": "ok"
```

sivassrr, 2017-08-10:
May be this issue need to go to Skiboot, if BMC is not having any role on this reboot.

sivassrr, 2017-08-10:
Looks like related to host checkstop condition detected. 
```
Aug 03 15:59:29 witherspoon ipmid[1127]: fru data size:[256], area offset:[8], area_size:[248]
Aug 03 15:59:29 witherspoon ipmid[1127]: Successfully verified area checksum. offset:[8]
Aug 03 15:59:29 witherspoon ipmid[1127]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru1a]
Aug 03 15:59:29 witherspoon ipmid[1127]: ERROR Unable to get the fru info for FRU=26
Aug 03 15:59:29 witherspoon ipmid[1127]: Error updating inventory
Aug 03 15:59:29 witherspoon ipmid[1127]: A host system event was received
Aug 03 15:59:29 witherspoon dbus[953]: [system] Activating via systemd: service name='org.freedesktop.journal1' unit='dbus-org.freedesktop.journal1.service'
Aug 03 15:59:29 witherspoon systemd[1]: Starting Journal Dbus Service...
Aug 03 15:59:30 witherspoon dbus[953]: [system] Successfully activated service 'org.freedesktop.journal1'
Aug 03 15:59:30 witherspoon systemd[1]: Started Journal Dbus Service.
Aug 03 15:59:30 witherspoon phosphor-fru-fault-monitor[1089]: /xyz/openbmc_project/logging/entry/19
Aug 03 15:59:30 witherspoon phosphor-ledmanager[952]: Asserting LEDs
Aug 03 15:59:31 witherspoon ipmid[1127]: A host system event was received
Aug 03 15:59:32 witherspoon phosphor-fru-fault-monitor[1089]: /xyz/openbmc_project/logging/entry/20
Aug 03 15:59:32 witherspoon phosphor-gpio-monitor[1287]: GPIO line altered
Aug 03 15:59:32 witherspoon systemd[1]: Created slice system-openpower\x2ddebug\x2dcollector\x2dcheckstop.slice.
Aug 03 15:59:32 witherspoon systemd[1]: Stopping Phosphor GPIO checkstop monitor...
Aug 03 15:59:32 witherspoon systemd[1]: Starting OpenPOWER debug data collector for host checkstop...
Aug 03 15:59:32 witherspoon systemd[1]: Stopped Phosphor GPIO checkstop monitor.
Aug 03 15:59:38 witherspoon checkstop_app[1620]: Host checkstop condition detected
Aug 03 15:59:38 witherspoon phosphor-fru-fault-monitor[1089]: /xyz/openbmc_project/logging/entry/21
Aug 03 15:59:38 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Aug 03 15:59:38 witherspoon systemd[1]: Reached target Quiesce Target.
```

```
Aug 03 16:01:46 witherspoon phosphor-gpio-monitor[1691]: GPIO line altered
Aug 03 16:01:46 witherspoon systemd[1]: Stopping Phosphor GPIO checkstop monitor...
Aug 03 16:01:46 witherspoon systemd[1]: Starting OpenPOWER debug data collector for host checkstop...
Aug 03 16:01:46 witherspoon systemd[1]: Stopped Phosphor GPIO checkstop monitor.
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]: Traceback (most recent call last):
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:     result = self._run(*self.args, **self.kwargs)
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:     ciphers=ciphers)
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Aug 03 16:01:46 witherspoon phosphor-gevent[2048]:     self.do_handshake()
Aug 03 16:01:47 witherspoon phosphor-gevent[2048]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Aug 03 16:01:47 witherspoon phosphor-gevent[2048]:     return self._sslobj.do_handshake()
Aug 03 16:01:47 witherspoon phosphor-gevent[2048]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Aug 03 16:01:47 witherspoon phosphor-gevent[2048]: <Greenlet at 0x75fc53f0: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x75f9bb50 fileno=5 address=('::34f2:ea76:b0cd:6076%1986055584', 443, 0, 1986055584)>>(<socket at 0x75e72830 fileno=9>, ('::80d9:5e76:8e91:f402%1995037468', 44518, 0, 199)> failed with SSLEOFError
Aug 03 16:01:47 witherspoon phosphor-gevent[2048]: ::80d9:5e76:8e91:f402%1995037468 - - [2017-08-03 16:01:47] "POST /login HTTP/1.1" 200 320 0.061819
Aug 03 16:01:51 witherspoon checkstop_app[2057]: Host checkstop condition detected
Aug 03 16:01:51 witherspoon phosphor-fru-fault-monitor[1089]: /xyz/openbmc_project/logging/entry/24
Aug 03 16:01:51 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Aug 03 16:01:51 witherspoon systemd[1]: Reached target Quiesce Target.
Aug 03 16:01:52 witherspoon systemd[1]: Reached target Host instance 0 crashed.
Aug 03 16:01:52 witherspoon phosphor-host-state-manager[1164]: Auto reboot enabled. Reducing HOST BOOTCOUNT by 1.
Aug 03 16:01:52 witherspoon phosphor-host-state-manager[1164]: Beginning reboot...
Aug 03 16:01:52 witherspoon phosphor-host-state-manager[1164]: Host State transaction request
Aug 03 16:01:52 witherspoon systemd[1]: Starting Soft power off of the hos
```

gkeishin, 2017-08-10:
This one is expected if we are looking at the Host going quiesce , doing few reboots to poweron.. Like @sivassrr pointed out there is a checkstop coming down triggering the Host reboot sequence.

So if we need to track this , it needs to be why and where the Checkstop comes from and it will need the SOL output and esel to debug I would say

rahulmah, 2017-08-11:
This is where it is rebooting from SOL.

```
 85.32883|------------------------------------------------
 85.32884|  Callout type             : Procedure Callout
 85.32885|  Procedure                : EPUB_PRC_HB_CODE
 85.32885|  Priority                 : SRCI_PRIORITY_HIGH
 85.32886|------------------------------------------------
 85.32887|  host_runtime_setup
 85.32888|------------------------------------------------
 85.32889|  Hostboot Build ID: hostboot-faac61e/hbicore.bin
 85.32889|================================================

--== Welcome to Hostboot hostboot-faac61e/hbicore.bin ==--

  4.05097|secure|Booting in non-secure mode.
  4.05099|secure|SecureROM invalid - skipping functionality
  4.69572|ERRL|Dumping errors reported prior to registration
  5.24604|ISTEP  6. 5 - host_init_fsi
  5.32319|ISTEP  6. 6 - host_set_ipl_parms
  5.34209|ISTEP  6. 7 - host_discover_targets
  5.79479|HWAS|PRESENT> DIMM[03]=FFFF000000000000
  5.79480|HWAS|PRESENT> Proc[05]=C000000000000000
  5.79481|HWAS|PRESENT> TPM[31]=8000000000000000
  5.82171|ISTEP  6. 8 - host_update_master_tpm
  6.01233|SECURE|Security Access Bit> 0x0000000000000000
  6.01234|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
  6.01257|ISTEP  6. 9 - host_gard
  6.05981|ISTEP  6.10 - host_revert_sbe_mcs_setup
```

rahulmah, 2017-08-11:
Okay. I will move this to skiboot.

gkeishin, 2017-08-11:
Closing this one.. and add the skiboot giithub# once done.

closed 2017-08-11 by gkeishin