openbmc/openbmc #2134: BMC: Mount RW filesystem and overlay

anoo1, 2017-08-10:


cbostic, 2017-08-10:
++++++++++++++++++++++++++++++++++++++
TASK LIST

- [ ] Code up to mount tmpfs on /var  via fstab
- [ ] Verify proper mount
- [ ] Identify proper location to mount overlay on /etc   fstab, systemd, script?
- [ ] Verify /etc can be written to in overlay mode

mdmillerii, 2017-08-10:
I thought we were going to mount `/var` and `/etc` via script not fstab.  We are already violating [systemd rules](https://www.freedesktop.org/wiki/Software/systemd/FileHierarchy/) by not having the real `/etc` when the [generators are run](https://www.freedesktop.org/wiki/Software/systemd/Generators/).  In addition to the `systemd-fstab-generator` parsing `/etc/fstab` to generate a systemd unit file, the generators are running with a read-only stale `/etc`.  

At a minimum if we do not enable [`autofs4`](https://lists.ozlabs.org/pipermail/openbmc/2017-July/008381.html) we will need to make sure we don't [ask for automounts](https://www.freedesktop.org/software/systemd/man/systemd-mount.html#)

I feel we should not rely on generators to mount `/etc`.

eddiejames, 2017-08-12:
# **STATUS**

08/11 - Got /var mounted on tmpfs.

rfrandse, 2017-08-24:
<https://gerrit.openbmc-project.xyz/5962> filesystem: mount rw and overlay
Resolves  #2134 BMC: Mount RW filesystem and overlay

closed 2017-08-29 by williamspatrick