openbmc/openbmc #2137: Re-create BMC Software Version after reboot

saqibkh, 2017-08-11:
The dbus objects for all BMC Active images but be recreated after a BMC is rebooted including the priority

saqibkh, 2017-08-11:
- [ ]  Create a service that mounts the RO ubi volume for BMC images when a system boots up.
- [ ]  Recreate the dbus objects for BMC Software versions by checking /media for available versions.
- [ ]  Restore the priority by reading the persistence priority files.

saqibkh, 2017-08-11:
Sequence of commits:
1) https://gerrit.openbmc-project.xyz/#/c/5934/
2) https://gerrit.openbmc-project.xyz/#/c/5946/

rfrandse, 2017-08-11:
<https://gerrit.openbmc-project.xyz/5946> BMC: Restore version and activation dbus objects on BMC reboot.
Resolves  #2137 Re-create BMC Software Versions after reboot

closed 2017-08-30 by williamspatrick