openbmc/openbmc #2138: IPMI:- Implement Network mode change support from IPMI

ratagupt, 2017-08-11:
User should be able to change the interface mode(Static/Dynamic) from IPMI

ratagupt, 2017-08-11:
Confirmed with @whowutwut(victor hu) that we don't need this support in phase5.

whowutwut, 2017-08-11:
Thanks @ratagupt .  

@zet809, FYI this was a request for inband change for ipsrc to dynamic, I searched our code and we don't call that.. so I think it's not needed.

ratagupt, 2017-08-12:
Here we need to support multiple interfaces through IPMI,Currently IPMI supports single interface.

ratagupt, 2017-08-23:
I had a discussion with @whowutwut 
and he confirmed that we  need to cover the following:

Scenario 1
1) Factory Reset
2) Get DHCP IP  
3) ipmitool to change network properties to static

Here after step 3 we need to change the mode to static.so we need this in phase 5.

Scenario 2:   (broke with new code)
1) factory reset
2) In DHCP mode
3) 10pin direct to BMC and run

ratagupt, 2017-08-24:
- [ ] Implement the support for following command.
        ipmitool lan set 1 ipsrc static/dynamic.

Will list out the various condition and verify it with Victor Hu, that how it is implemented in other stack.

whowutwut, 2017-08-24:
@ratagupt  This is an implementation to support this via REST?  What is the proposed endpoint for it?

ratagupt, 2017-08-25:
@whowutwut :- Suppose the system ethenet interface is in DHCP  mode and through IPMI you want to set some ip through the below commands.

Currently IPMI doesn't support the ipsrc static/dynamic command so you would not be able to change the mode of the ethernet interface hence the IP would not be added.

However if the interface mode is Static then it will work fine.

Now with the implementation of ipsrc command type through IPMI will have below behavior.

==>If eth0 is having DHCP IP and through IPMI user change the mode to SATTIC then user have to supply all the data(address/mask/gateway).we will not use the IP which is already assigned to eth0 through DHCP.

==>If eth0 is having DHCP IP and through IPMI user change the mode to DHCP,it is a NO-OP

==>if eth0 is having Static IP and through IPMI user says that he wants to enable DHCP then it may happen that interface gets new ip address from DHCP server.

==>if eth0 is having Static IP and through IPMI user change the mode to static then we will inherit the ip info which is configured on the system if user has given partial data.

whowutwut, 2017-08-25:
@zet809 FYI above

whowutwut, 2017-08-30:
@ratagupt  Is the above behavior default behavior? What is suppose to happen in RedFish specification?  Really, we just want the BMC to behave in a documented fashion, if you document this behavior on the README, that is fine.  Otherwise how would other consumers of the API know the intended behavior without actually trying it out to see what happens...

ratagupt, 2017-08-30:
@whowutwut :- Every protocol comes with its own limitation..But seems to me the behaviour would be same. except last one,User has to give all the data, all this limittaion comes as IPMI supports either tagged packet or untagged packet..That limittaion I don't see in REST/REDFISH.
Regarding this behaviour I would update the README.

ratagupt, 2017-09-01:
# **STATUS**
08/31
Starting the coding today.
09/01
Still Coding,Planning it to complete it by today.

rfrandse, 2017-09-03:
<https://gerrit.openbmc-project.xyz/6383> SetLan: Make changes in set channel access to support ipsrc param
Resolves  #2138 IPMI:- Implement Network mode change support from IPMI

closed 2017-09-08 by williamspatrick