openbmc/openbmc #2139: Witherspoon : Host and chassis state remains off after powering on Host (Inconsistent)

rahulmah, 2017-08-11:
Some time we are seeing that, Host and chassis state remains off after powering on Host. We dont have the exact step to reproduce the same. See if the attached logs help.

FFDC : 
[Host and chassis state unchnage on power on.zip](https://github.com/openbmc/openbmc/files/1218433/Host.and.chassis.state.unchnage.on.power.on.zip)

Firmware info: 
bash-4.1$ curl -c cjar -b cjar -k https://wsbmc006/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/136cf504": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.17-210-gbabb528,buildroot-2017.05-93-g9b91216,skiboot-v5.7,hostboot-f523eaa,linux-4.12.1-openpower1-p8927fdf,petitboot-v1.5.1-pce8abef,machine-xml-5575e7e,occ-ca84830,hostboot-binaries-c24fccd,capp-ucode-9c73e9f,sbe-02021c6-p6a0f00f",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.87"
    },
    "/xyz/openbmc_project/software/5e9b20c5": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-216-gacab2a0"
    }
  },
  "message": "200 OK",
  "status": "ok"

geissonator, 2017-08-11:
The data you provided above is for the PNOR software states?  The issue implies it has to do with host and chassis states though, not the software activation levels?

rahulmah, 2017-08-12:
Please check "chassis state unchnage on power on.zip"  file for all the FFDC data (e.g. journal log, dmesg etc.).  Software data is for the firmware of from the system.

geissonator, 2017-08-16:
Please provide recreate directions and show the system in the incorrect state (obmcutil state).  Just asking us to parse some ffdc and look for an issue is not very useful, or a good use of time.

closed 2017-08-16 by geissonator

sivassrr, 2017-08-16:
```
Aug 03 15:45:40 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Aug 03 15:45:40 witherspoon kernel[1028]: [  173.620000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Aug 03 15:45:41 witherspoon systemd[1]: Started Turn VCS on.
Aug 03 15:45:41 witherspoon systemd[1]: Starting Power on bind ir35221 device driver devs...
Aug 03 15:45:41 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Aug 03 15:45:41 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Aug 03 15:45:41 witherspoon systemd[1]: Started Power on bind ir35221 device driver devs.
Aug 03 15:45:41 witherspoon systemd[1]: Reached target Chassis0 (Power On).
Aug 03 15:45:41 witherspoon systemd[1]: Reached target Start Host0 (Pre).
Aug 03 15:45:41 witherspoon systemd[1]: Starting Start host0 SBE...
Aug 03 15:45:41 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Aug 03 15:45:42 witherspoon phosphor-chassis-state-manager[1222]: Received signal that power ON is complete
Aug 03 15:45:42 witherspoon phosphor-chassis-state-manager[1222]: Change to Chassis Power State
```  
Not able to see anything weird here in journal. Power on happening and see Chassis power state is  getting changed. But how can we verify that  phosphor-chassis-state-manager update it correctly the state.