openbmc/openbmc #2140: [xCAT] Reseting the BMC to DHCP causes issues with REST Service and IPv6 when coming back up

whowutwut, 2017-08-12:
With issue https://github.com/openbmc/openbmc/issues/1952 We are able to get a REST command to clear the network to a "factory" default with DHCP enabled.... However, running this causes some problems..  

@ratagupt @williamspatrick  FYI... 

I've run this on the following firmware: 
```
ibm-v1.99.8-188-g91af6d-test0-0-g7ea721b-dirty
IBM-witherspoon-redbud-ibm-OP9_v1.18_1.19
```

Here's the command I used to reset the bmc, based on your #1952... and it works..  the `00-bmc-eth0.network` file is cleared out... 
```
# ./reset_dhcp.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data": [] }' https://10.6.5.254/xyz/openbmc_project/network/action/Reset
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

The first time I ran it, the MAC address got hosed on the node... 
```
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr D6:26:21:9B:79:23
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::d426:21ff:fe9b:7923%1112880016/64 Scope:Link
``` 
I could not re-create this the 2nd time so not sure what happened... but I fixed it by running: `fw_setenv ethaddr 70:E2:84:14:28:0D` and rebooting the BMC... 

On the 2nd time...  I knew more about what to do.... here's what happened 

--------------

###  Before, take a look at the BMC config 

```
root@witherspoon:~# cd /etc/systemd/network/
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0
[Network]
Address=10.6.5.254/8
Gateway=10.6.29.1
root@witherspoon:/etc/systemd/network# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::72e2:84ff:fe14:280d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5629 errors:0 dropped:0 overruns:0 frame:0
          TX packets:547 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:644607 (629.4 KiB)  TX bytes:52351 (51.1 KiB)
          Interrupt:20
```

### Reset 

```
[root@stratton01 frame6u05]# pwd
/root/c910env/notes/openbmc/UT/Network/frame6u05

[root@stratton01 frame6u05]# ./login.sh
==> IP=10.6.5.254
+ curl -c cjar -k -X POST -H 'Content-Type: application/json' -d '{"data": [ "root", "0penBmc" ] }' https://10.6.5.254/login
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}

[root@stratton01 frame6u05]# ./reset_dhcp.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data": [] }' https://10.6.5.254/xyz/openbmc_project/network/action/Reset
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
```

Look in the BMC....

```
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0
[Network]
DHCP=true
[DHCP]
ClientIdentifier=mac
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::72e2:84ff:fe14:280d%1112880016/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6070 errors:0 dropped:0 overruns:0 frame:0
          TX packets:594 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:695078 (678.7 KiB)  TX bytes:58672 (57.2 KiB)
          Interrupt:20

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1112880016/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2724 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2724 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:207248 (202.3 KiB)  TX bytes:207248 (202.3 KiB)
```

### Reboot BMC

```
[root@stratton01 frame6u05]# ./reboot_bmc.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X PUT -d '{"data": "xyz.openbmc_project.State.BMC.Transition.Reboot" }' https://10.6.5.254/xyz/openbmc_project/state/bmc0/attr/RequestedBMCTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
```

### After BMC reboots, check out the MAC address

```
root@witherspoon:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet6 addr: fe80::72e2:84ff:fe14:280d%1112880016/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:302 errors:0 dropped:0 overruns:0 frame:0
          TX packets:124 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:35854 (35.0 KiB)  TX bytes:13147 (12.8 KiB)
          Interrupt:20

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1112880016/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:240 errors:0 dropped:0 overruns:0 frame:0
          TX packets:240 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:18240 (17.8 KiB)  TX bytes:18240 (17.8 KiB)

root@witherspoon:~#
```

Seems OK this time... `fw_printenv` seems to have the MAC....

```
root@witherspoon:~# fw_printenv
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=fdt addr 20080000; if fdt get value ramdisk_conf /configurations/conf@1 ramdisk; then     bootm 20080000; else bootm 20080000 20300000; fi
bootdelay=2
baudrate=115200
verify=yes
spi_dma=yes
ethaddr=70:E2:84:14:28:0D
```

### Set IP address static 

Run: 
```
busctl call org.openbmc.NetworkManager /org/openbmc/NetworkManager/Interface org.openbmc.NetworkManager SetAddress4 ssss eth0 10.6.5.254 255.0.0.0 10.6.29.1
```

```
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1363 errors:0 dropped:0 overruns:0 frame:0
          TX packets:217 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:160052 (156.3 KiB)  TX bytes:24225 (23.6 KiB)
          Interrupt:20
```

## ISSUE 1, IPV6 inet6 is gone 

After I set the IPv4 Address, the IPv6 inet6 is gone, a reboot resolves this though and it's back...

```
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::72e2:84ff:fe14:280d%1112880016/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13843 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1319 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1596808 (1.5 MiB)  TX bytes:121995 (119.1 KiB)
          Interrupt:20
```

## ISSUE 2, REST calls do not work...

In this configuration, I'm unable to call any REST calls, like the LOGIN
```
root@witherspoon:~# uptime
 02:16:06 up 5 min,  load average: 1.24, 2.30, 1.22

[root@stratton01 frame6u05]# ./login.sh
==> IP=10.6.5.254
+ curl -c cjar -k -X POST -H 'Content-Type: application/json' -d '{"data": [ "root", "0penBmc" ] }' https://10.6.5.254/login
curl: (7) Failed connect to 10.6.5.254:443; Connection refused
```

To resolve this, I have to reboot the BMC...   it response and takes a long time for the first one but it responds and works...
```
[root@stratton01 frame6u05]# ./login.sh
==> IP=10.6.5.254
+ curl -c cjar -k -X POST -H 'Content-Type: application/json' -d '{"data": [ "root", "0penBmc" ] }' https://10.6.5.254/login
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
```

So but in the past, setting the IP address did NOT require a reboot for the REST services to work.. so something else must have changed..

ratagupt, 2017-08-14:
@whowutwut :- I did see that you added the static IP with the old org interface so it was not related to xyz.
busctl call org.openbmc.NetworkManager /org/openbmc/NetworkManager/Interface org.openbmc.NetworkManager SetAddress4 ssss eth0 10.6.5.254 255.0.0.0 10.6.29.1.

whowutwut, 2017-08-14:
Thanks @ratagupt ! Will try this out tomorrow and update this issue