openbmc/openbmc #2145: UBIFS error (pid: 1033): cannot open "ubi7:pnor-patch", error -19 is seen in dmesg

PavamanSubramaniyam, 2017-08-14:
After upgrading the witherspoon machine to the latest master Open BMC build, we can see the following error  messages in dmesg.

```
[   25.890000] systemd-journald[602]: Received request to flush runtime journal from PID 1
[   39.030000] ubi7: default fastmap pool size: 100
[   39.030000] ubi7: default fastmap WL pool size: 50
[   39.030000] ubi7: attaching mtd7
[   39.030000] ubi7: scanning is finished
[   39.030000] ubi7 error: ubi_read_volume_table: the layout volume was not found
[   39.030000] ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
[   49.900000] 8021q: adding VLAN 0 to HW filter on device eth0
[   53.210000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 1
[   53.210000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 2
[   53.210000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[   53.210000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[   53.800000] UBIFS error (pid: 1033): cannot open "ubi7:pnor-patch", error -19
[  260.750000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  273.850000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[  276.440000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  278.740000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[  332.040000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  340.760000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[  342.470000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  343.910000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[  396.100000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  407.200000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[  409.550000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  411.560000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[  464.770000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  477.180000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[  479.260000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[  481.180000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[ 2347.950000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2358.700000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 2360.870000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2362.680000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[ 2414.020000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2424.850000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 2427.280000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2429.180000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[ 2480.130000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2489.240000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 2491.480000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2492.730000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
[ 2544.440000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2552.900000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 2555.590000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 2558.080000]  slave@00:00: status: 0xc4000511, sisc: 0x82040000
```

The machine is installed with the latest Open BMC Firmware image.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-231"
VERSION_ID="v1.99.8-231-gdf827b6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-231"
BUILD_ID="v1.99.8"
```

PavamanSubramaniyam, 2017-08-14:
[20170814062837038020_Myffdc.zip](https://github.com/openbmc/openbmc/files/1221411/20170814062837038020_Myffdc.zip)

geissonator, 2017-08-22:
Covered in a different bug @mtritz ?

mdmillerii, 2017-08-22:
-19 is `-ENODEV`, no such device.  Above in the error log we see the ubi attach failed:
```
[   39.030000] ubi7: attaching mtd7
[   39.030000] ubi7: scanning is finished
[   39.030000] ubi7 error: ubi_read_volume_table: the layout volume was not found
[   39.030000] ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
```
before the headline error
```
[   53.800000] UBIFS error (pid: 1033): cannot open "ubi7:pnor-patch", error -19
```

What was the history of the machine?

anoo1, 2017-08-23:
Also wondering if the pnor is formatted as ubi. We're adding a check to pnor patch to check before attempting the attach. Has this system being updated with a squashfs.pnor file?

mtritz, 2017-08-24:
# **STATUS**
8/24: We suspect that this commit https://gerrit.openbmc-project.xyz/#/c/6095/ will resolve this issue. Unless we receive information indicating otherwise from @PavamanSubramaniyam, the issue will be treated as resolved.

PavamanSubramaniyam, 2017-08-28:
I have installed the latest Open BMC image from master tree.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-57"
VERSION_ID="v1.99.9-57-g8aa1141"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-57"
BUILD_ID="v1.99.9"

[   25.480000] ubi7: default fastmap pool size: 100
[   25.480000] ubi7: default fastmap WL pool size: 50
[   25.480000] ubi7: attaching mtd7
[   25.720000] ubi7: scanning is finished
[   25.720000] ubi7: attached mtd7 (name "pnor", size 128 MiB)
[   25.780000] ubi7: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[   25.780000] ubi7: min./max. I/O unit sizes: 1/256, sub-page size 1
[   25.780000] ubi7: VID header offset: 64 (aligned 64), data offset: 128
[   25.780000] ubi7: good PEBs: 2048, bad PEBs: 0, corrupted PEBs: 0
[   25.780000] ubi7: user volume: 6, internal volumes: 1, max. volumes count: 128
[   25.780000] ubi7: max/mean erase counter: 3/1, WL threshold: 4096, image sequence number: 1835199429
[   25.780000] ubi7: available PEBs: 468, total reserved PEBs: 1580, PEBs reserved for bad PEB handling: 0
[   25.780000] ubi7: background thread "ubi_bgt7d" started, PID 1009
[   26.830000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 1
[   26.830000] ftgmac100 1e660000.ethernet eth0: ncsi: removed vlan tag 0 at index 2
[   26.830000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[   26.830000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[   35.130000] ubi: mtd7 is already attached to ubi7
[   47.250000] block ubiblock7_0: created from ubi7:0(pnor-ro-39885ab3)
[   50.010000] UBIFS (ubi7:1): UBIFS: mounted UBI device 7, volume 1, name "pnor-rw-39885ab3"
[   50.010000] UBIFS (ubi7:1): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   50.010000] UBIFS (ubi7:1): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   50.010000] UBIFS (ubi7:1): reserved for root: 763076 bytes (745 KiB)
[   50.010000] UBIFS (ubi7:1): media format: w5/r0 (latest is w5/r0), UUID 23BEAD76-2AD2-4445-AE51-D097B8052E0B, small LPT model
[   50.010000] UBIFS (ubi7:1): background thread "ubifs_bgt7_1" started, PID 1110
[   50.010000] UBIFS (ubi7:2): UBIFS: mounted UBI device 7, volume 2, name "pnor-prsv"
[   50.010000] UBIFS (ubi7:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   50.010000] UBIFS (ubi7:2): FS size: 1504384 bytes (1 MiB, 23 LEBs), journal size 523265 bytes (0 MiB, 6 LEBs)
[   50.010000] UBIFS (ubi7:2): reserved for root: 71055 bytes (69 KiB)
[   50.010000] UBIFS (ubi7:2): media format: w5/r0 (latest is w5/r0), UUID B4101322-CB42-4364-A44E-0DC88098CA02, small LPT model
[   50.010000] UBIFS (ubi7:2): background thread "ubifs_bgt7_2" started, PID 1117
[   52.290000] UBIFS (ubi7:4): background thread "ubifs_bgt7_4" started, PID 1130
[   52.290000] UBIFS (ubi7:4): UBIFS: mounted UBI device 7, volume 4, name "pnor-rw-6124ea64"
[   52.290000] UBIFS (ubi7:4): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   52.290000] UBIFS (ubi7:4): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   52.290000] UBIFS (ubi7:4): reserved for root: 763076 bytes (745 KiB)
[   52.290000] UBIFS (ubi7:4): media format: w5/r0 (latest is w5/r0), UUID A0B942C7-6CFB-4033-B459-0E2EC40F828F, small LPT model
[   55.920000] block ubiblock7_5: created from ubi7:5(pnor-ro-6124ea64)
[   57.470000] UBIFS (ubi7:3): UBIFS: mounted UBI device 7, volume 3, name "pnor-patch"
[   57.470000] UBIFS (ubi7:3): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   57.470000] UBIFS (ubi7:3): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
[   57.470000] UBIFS (ubi7:3): reserved for root: 763076 bytes (745 KiB)
[   57.470000] UBIFS (ubi7:3): media format: w5/r0 (latest is w5/r0), UUID B512EF22-892F-4471-8A19-A8F117436F2E, small LPT model
[   57.670000] UBIFS (ubi7:3): background thread "ubifs_bgt7_3" started, PID 1162
```

The dmesg output doesn't show the UBIFS error any more.

closed 2017-08-28 by geissonator