openbmc/openbmc #2150: Network: VLAN: vlan ID is shown even after deleting it

prkatti1, 2017-08-15:
Steps:
1. Create VLAN
2. Add IP address
3. Delete IP
4. Delete VLAN
5. Try to enumerate deleted VLAN, it gets 200OK message instead of error message.

}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/config": {
      "DefaultGateway": "9.3.20.1",
      "HostName": "witherspoon"
    },
    "/xyz/openbmc_project/network/config/dhcp": {
      "DNSEnabled": 0,
      "HostNameEnabled": 0,
      "NTPEnabled": 0
    },
    "/xyz/openbmc_project/network/eth0": {
      "AutoNeg": 0,
      "DHCPEnabled": 0,
      "DomainName": [],
      "InterfaceName": "eth0",
      "MACAddress": "70:e2:84:14:28:76",
      "Nameservers": [],
      "Speed": 0
    },
    "/xyz/openbmc_project/network/eth0/ipv4/e9767624": {
      "Address": "9.3.21.42",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 23,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0/ipv6/913ae752": {
      "Address": "fe80::72e2:84ff:fe14:2876",
      "Gateway": "",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.LinkLocal",
      "PrefixLength": 64,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv6"
    },
    "/xyz/openbmc_project/network/eth0_40/ipv4/d64ba716": {
      "Address": "10.6.6.5",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 24,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0_50": {
      "AutoNeg": 0,
      "DHCPEnabled": 0,
      "DomainName": [],
      "Id": 50,
      "InterfaceName": "eth0.50",
      "MACAddress": "70:e2:84:14:28:76",
      "Nameservers": [],
      "Speed": 0
    },
    "/xyz/openbmc_project/network/host0/intf": {
      "MACAddress": "00:00:00:00:00:00"
    },
    "/xyz/openbmc_project/network/host0/intf/addr": {
      "Address": "0.0.0.0",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 0,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    }
  },
  "message": "200 OK",
  "status": "ok"
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X DELETE https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/eth0_50
curl: (52) Empty reply from server
(reverse-i-search)`L': curl -c cjar -b cjar -k -H "Content-Type: application/json" -X DE^CTE https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/eth0_50
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://wsbmc011.aus.stglabs.ibm.com/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -XGET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/eth0_50/enumerate
{
  "data": {},
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -XGET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/enumerate
{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException(\"Unknown interface 'xyz.openbmc_project.Network.MACAddress'.\",)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 800, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 857, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1040, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1014, in wrap",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 979, in wrap",
      "    data = callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 200, in __call__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 306, in do_get",
      "    mapper_data=request.route_data['map']).dataitems()}",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 188, in enumerate_subtree",
      "    path, owner, interfaces, match))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 111, in __get_properties_on_bus",
      "    properties_iface, i))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 97, in __get_properties_on_iface",
      "    properties_iface.GetAll, iface)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 83, in __try_properties_interface",
      "    return f(*a)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'xyz.openbmc_project.Network.MACAddress'."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/
{
  "data": [
    "/xyz/openbmc_project/network/config",
    "/xyz/openbmc_project/network/eth0_50",
    "/xyz/openbmc_project/network/host0",
    "/xyz/openbmc_project/network/eth0_40",
    "/xyz/openbmc_project/network/eth0"
  ],
  "message": "200 OK",
  "status": "ok"

ratagupt, 2017-08-16:
I think they both are same..
https://github.com/openbmc/openbmc/issues/2152.
Please do the same as I mentioned in 2152,check with your finding with busctl.

ratagupt, 2017-08-17:
The problem is weird.
busctl works fine but seems object mapper is having some problem that is the reason through REST there is inconsistent behavior.
Here is the summary of the defect what is happening?

Create the VLAN Interface
========================
busctl call  xyz.openbmc_project.Network /xyz/openbmc_project/network xyz.openbmc_project.Network.VLAN.Create VLAN su "eth0" 50

Delete the VLAN Interface
======================
busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0_50 xyz.openbmc_project.Object.Delete Delete

Ask Mapper to get all the objects under /xyz/openbmc_project/network
===========================================================
busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetSubTree sias "/xyz/openbmc_project/network" 0 1 "xyz.openbmc_project.Network.EthernetInterface"

a{sa{sas}} 2 "/xyz/openbmc_project/network/eth0_50" 1 ":1.247" 1 "xyz.openbmc_project.Network.EthernetInterface" "/xyz/openbmc_project/network/eth0" 1 ":1.247" 1 "xyz.openbmc_project.Network.EthernetInterface"

We got the eth0_50 here,that is the problem.
====================================

Now I will look into busctl..This is good.I don't see the eth0_50 there.
======================================================
busctl tree xyz.openbmc_project.Network
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/network
      ├─/xyz/openbmc_project/network/config
      │ └─/xyz/openbmc_project/network/config/dhcp
      └─/xyz/openbmc_project/network/eth0
        ├─/xyz/openbmc_project/network/eth0/ipv4
        │ └─/xyz/openbmc_project/network/eth0/ipv4/880757f
        └─/xyz/openbmc_project/network/eth0/ipv6
          └─/xyz/openbmc_project/network/eth0/ipv6/3b4f7d87

I took a look through busctl monitor,I can see that Interfaces removed signal was coming.
I found one problem which is in the network_manager_main.cpp,
We need to request the bus name after creation of object manager object.

ratagupt, 2017-08-18:
https://gerrit.openbmc-project.xyz/#/c/6078/2

rfrandse, 2017-08-18:
<https://gerrit.openbmc-project.xyz/6078> vlan:Remove the multiple implementation of sdbusplus object
Resolves  #2150 Network: VLAN: vlan ID is shown even after deleting it

ratagupt, 2017-08-18:
@prkatti1 : has verified the fix and it works fine in his testing.

closed 2017-08-23 by williamspatrick