openbmc/openbmc #2151: Network:VLAN: BMC does not throw when we try to add existing VLAN

prkatti1, 2017-08-15:
BMC does not throw when we try to add existing VLAN

1. Create VLAN 40
2. Add IP address to vlan 40
3. Now add another vlan with same ID 40
4. It does not throw any error and n/w module restarts
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -XGET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/eth0_40/
{
  "data": [
    "/xyz/openbmc_project/network/eth0_40/ipv4"
  ],
  "message": "200 OK",
  "status": "ok"
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["eth0",40] }' https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/action/VLAN               {
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["eth0",40] }' https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/action/VLAN
curl: (7) couldn't connect to host
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://wsbmc011.aus.stglabs.ibm.com/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -XPOST -d '{"data": ["eth0",40] }' https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/action/VLAN
curl: (52) Empty reply from server

ratagupt, 2017-08-16:
Currently we don't have support to throw the error back to the client so you can ignore this.

prkatti1, 2017-08-16:
ok @ratagupt . @sivassrr seems like this support is not there now.

ratagupt, 2017-08-18:
This can be taken through network validation story.

closed 2019-04-26 by stale[bot]