openbmc/openbmc #2152: Network: VLAN: vlan listing through REST and on system is different

prkatti1, 2017-08-15:
Steps:
1. Created 2 VLANs 50 and 40
2. Delete VLAN 50
3. Delete VLAN 40's IP address
4. Enumereate network, it does not show vlan 40

}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://xx.xx.xx.xx/xyz/openbmc_project/network/eth0_40/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/eth0_40/ipv4/d64ba716": {
      "Address": "10.6.6.5",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 24,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    }
  },
  "message": "200 OK",
  "status": "ok"

bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X DELETE https://xx.xxx.xx.xx/xyz/openbmc_project/network/eth0_40/ipv4/d64ba716
curl: (52) Empty reply from server

bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["xyz.openbmc_project.Network.IP.Protocol.IPv4","10.6.6.7",24,"0.0.0.0"] }' https://xx.xx.xx.xx/xyz/openbmc_project/network/eth0_40/action/IP
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/network/eth0_40"
  },
  "message": "404 Not Found",
  "status": "error"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -XGET https://xx.xx.xx.xx/xyz/openbmc_project/network/eth0_40/enumerate                                      {
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/network/eth0_40"
  },
  "message": "404 Not Found",
  "status": "error"

On witherspoon system:
root@witherspoon:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet xx.xx.x.xx/23 brd 9.3.21.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link
       valid_lft forever preferred_lft forever
4: eth0.40@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link
       valid_lft forever preferred_lft forever

ratagupt, 2017-08-16:
@prkatti1 : Can you update your findings with busctl also?
```busctl tree xyz.openbmc_project.Network" ```
Did you see the same behvaiour through biusctl?
It would be easy to narrow down the problem.

prkatti1, 2017-08-16:
2150 & 2152 are little different.
2150: We are deleting VLAN and enumerating it. Getting 200OK and it is getting listed via REST when we enumerate network
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://wsbmc011.aus.stglabs.ibm.com/xyz/openbmc_project/network/
{
"data": [
"/xyz/openbmc_project/network/config",
"/xyz/openbmc_project/network/eth0_50",
"/xyz/openbmc_project/network/host0",
"/xyz/openbmc_project/network/eth0_40",
"/xyz/openbmc_project/network/eth0"
],
"message": "200 OK",
"status": "ok"

2152: We are just deleting IP address from VLAN, vlan itself is not getting listed here.

dkodihal, 2017-08-17:
Probable dup of #2150. @ratagupt to confirm.

ratagupt, 2017-08-17:
Yes it is a dup of 2150.

ratagupt, 2017-08-18:
@dkodihal  can you please make dup of 2150?

rfrandse, 2017-08-18:
<https://gerrit.openbmc-project.xyz/6078> vlan:Remove the multiple implementation of sdbusplus object
Resolves  #2152 Network: VLAN: When we delete IP address from VLAN, that VLAN is not accessible & not getting listed via REST

ratagupt, 2017-08-18:
@prkatti1 : has verified the fix and it works fine in his testing.

closed 2017-08-23 by williamspatrick