openbmc/openbmc #2153: Network:VLAN: Able to send packets to and from LAN interface to VLAN interface

prkatti1, 2017-08-15:
1. Create VLAN 50
2. Add IP address to VLAN 50
3. Now ping VLAN 50 IP address from eth0 IP address
4. Do vice versa
5. In both scenarios ping is successful

root@witherspoon:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet xx.xx.xx.xx/23 brd 9.3.21.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link
       valid_lft forever preferred_lft forever
4: eth0.40@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link
       valid_lft forever preferred_lft forever
6: eth0.50@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue qlen 1000
    link/ether 70:e2:84:14:28:76 brd ff:ff:ff:ff:ff:ff
    inet 10.6.6.10/24 brd 10.6.6.255 scope global eth0.50
       valid_lft forever preferred_lft forever
    inet 10.6.6.9/24 brd 10.6.6.255 scope global secondary eth0.50
       valid_lft forever preferred_lft forever
    inet6 fe80::72e2:84ff:fe14:2876/64 scope link
       valid_lft forever preferred_lft forever
root@witherspoon:~# ping -I 10.6.6.9 xx.xx.xx.xx
PING xx.xx.xx.xx (xx.xx.xx.xx) from 10.6.6.9: 56 data bytes
64 bytes from xx.xx.xx.xx: seq=0 ttl=64 time=0.698 ms
64 bytes from xx.xx.xx.xx seq=1 ttl=64 time=0.363 ms
^C
--- 9.3.21.42 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.355/0.472/0.698 ms
root@witherspoon:~# ping -I xx.xx.xx.xx 10.6.6.9
PING 10.6.6.9 (10.6.6.9) from xx.xx.xx.xx: 56 data bytes
64 bytes from 10.6.6.9: seq=0 ttl=64 time=0.616 ms
64 bytes from 10.6.6.9: seq=1 ttl=64 time=0.407 ms
64 bytes from 10.6.6.9: seq=2 ttl=64 time=0.358 ms
^C
--- 10.6.6.9 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.358/0.460/0.616 ms

mdmillerii, 2017-08-15:
(1) Please use ` ``` ` before and after your logs to avoid markdown interpreting the text.
```
backtick quoting
```

(2) By default the Linux kernel will respond to arp requests for any interface on all interfaces.  See for example https://serverfault.com/questions/834512/why-does-linux-answer-to-arp-on-incorrect-interfaces .

There may be discussion required if setting the arp restrictions is desired.

ratagupt, 2017-08-16:
@prkatti1: As discussed and @mdmillerii has responded can you close this defect?

prkatti1, 2017-08-16:
Asi discussed with ratan & from the response from @mdmillerii, I am closing this issue.

closed 2017-08-16 by prkatti1