openbmc/openbmc #2154: Create power supply redundancy and derating factor sensors

geissonator, 2017-08-15:
Some details in #1213.  Need to implement these using the new ipmid sensor design.

geissonator, 2017-08-15:
Phase 5 for now, @spinler to check with occ team on impacts.

dhruvibm, 2017-08-24:
- [ ] Add YAMLs for ps derating and redundancy

- [ ] Create an application for handling PS redundancy and derating

- [ ] Add IPMI sensor read support for new interfaces.

tomjoseph83, 2017-09-01:
# **STATUS**

09/01
Made changes that are need for Phase 5. I would be completing 2220 first before moving on to this one.

tomjoseph83, 2017-09-07:
We should be removing the Power Supply redundancy/Power derating & Turbo allowed(Power specific) from the legacy dbus path check, once we get these changes in.

rfrandse, 2017-09-20:
<https://gerrit.openbmc-project.xyz/6782> Remove legacy reading of PS derating
Resolves  #2154 Create power supply redundancy and derating factor sensors

geissonator, 2017-11-29:
This was backed out, need to find story where we bring this back in.

dkodihal, 2017-11-30:
@dhruvibm can you use this story to write new dbus interfaces and app for the PS sensors, along with ipmid changes?

dhruvibm, 2017-12-14:
# **Status**
Need to decide the application implementing the PS derating and redundancy servers.

dkodihal, 2018-03-30:
This was implemented via https://gerrit.openbmc-project.xyz/#/q/topic:PowerSupplyRedundancy

tomjoseph83, 2018-05-21:
There are 2 sensors that needs to be moved from legacy code to sensor generated code.

a) Power Supply Redundancy is supported in the settings. So the input yaml for the sensor generation needs to be updated.

b) The implementation of power derating sensor needs to be discussed.

gtmills, 2018-07-18:
@tomjoseph83 @dkodihal What is this issue tracking now? Should it just be closed? We have a power supply redundancy sensor.

rfrandse, 2018-08-16:
moved to https://github.com/ibm-openbmc/dev/issues/66

closed 2018-08-16 by rfrandse

anoo1, 2018-09-12:
Done with:
https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/12165/
https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/12166/
https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/12167/