openbmc/openbmc #2156: Witherspoon front LEDs on solid at power off with latest build

bjwyman, 2017-08-15:
I am not quite certain what changed with the latest builds, but the LEDs on the front of a Witherspoon box end up in the on solid state at the power off state.

bjwyman, 2017-08-15:
With released build flashed onto system, after a BMC reset, at power off state:
```root@witherspoon:~# cat /etc/os-release
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.8"
VERSION_ID="ibm-v1.99.8-98-g06734f31-r2-0-g752ced3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.8"
BUILD_ID="ibm-v1.99.8-98-g06734f31-r2"
root@witherspoon:~#
...
root@witherspoon:/sys/class/leds/pca955x:front-id# ls
brightness      device          max_brightness  power           subsystem       trigger         uevent
root@witherspoon:/sys/class/leds/pca955x:front-id# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:front-id# cd ../pca955x\:front-fault
root@witherspoon:/sys/class/leds/pca955x:front-fault# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:front-fault# cd ../pca955x\:front-power
root@witherspoon:/sys/class/leds/pca955x:front-power# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:front-power# cd ../pca955x\:fan0
root@witherspoon:/sys/class/leds/pca955x:fan0# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:fan0# cd ../pca955x\:fan1
root@witherspoon:/sys/class/leds/pca955x:fan1# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:fan1# cd ../pca955x\:fan2
root@witherspoon:/sys/class/leds/pca955x:fan2# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:fan2# cd ../pca955x\:fan3
root@witherspoon:/sys/class/leds/pca955x:fan3# cat trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds/pca955x:fan3#
```
All LEDs were off (well, aside from the green inside the power button)

bjwyman, 2017-08-15:
After netbooting from:
https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/lastSuccessfulBuild/artifact/deploy/images/witherspoon/

Artifacts of ubuntu,witherspoon #977-5afd4d99063d18

```
root@witherspoon:~# date; uptime; cat /etc/os-release
Tue Aug 15 23:34:38 UTC 2017
 23:34:38 up 2 min,  load average: 2.78, 1.44, 0.55
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-253"
VERSION_ID="v1.99.8-253-g5afd4d9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-253"
BUILD_ID="v1.99.8"
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# cd /sys/class/leds
root@witherspoon:/sys/class/leds# ls
fan0         fan1         fan2         fan3         front-fault  front-id     front-power  rear-fault   rear-id      rear-power
root@witherspoon:/sys/class/leds# cat fan0/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat fan1/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat fan2/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat fan3/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat front-fault/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat front-id/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds# cat front-power/trigger
[none] timer heartbeat default-on
root@witherspoon:/sys/class/leds#
```
Changes
Git (git https://github.com/openbmc/openbmc.git)
Add fsidbg tool. Remove iicmaster. (commit: 8bcbb91) (detail)
PNOR: Create a service file for updating symlinks. (commit: fa92e97) (detail)
meta-mellanox: meta-msn: add msn specific defconfig layer. (commit: 5afd4d9) (detail)

All the front LEDs on solid (back the same as before).

vishwabmc, 2017-08-22:
@bjwyman Wonder if you looked at `cat brightness` ?, The value in `trigger` does not tell if its `solid-on`.

bjwyman, 2017-08-22:
I do not recall looking at the brightness value.

vishwabmc, 2017-08-22:
Looks like when the BMC is booted, the `brightness` is pre populated with `255`. To confirm this, I did these ;

1/. echo 0 > `/sys/class/leds/fan0/brightness`
2/. reboot
3/. Saw `255` in `/sys/class/leds/fan0/brightness`

bjwyman, 2017-08-22:
It likely did not do that with earlier builds.
BUILD_ID="ibm-v1.99.8-98-g06734f31-r2"
Pretty sure that did not do that. That should be the 1732A build/image.

vishwabmc, 2017-08-23:
I again tested this on latest master and saw that `255` was getting pre-populated into `brightness` as part of BMC boot.

@geissonator Please assign this to @shenki or someone from Linux

geissonator, 2017-08-23:
@amboar - can you take a look?

amboar, 2017-08-23:
Can someone please test this commit?

https://github.com/amboar/linux/commit/b948eeab06199e44fe8092e93afdfebcd2c183a1

vishwabmc, 2017-08-24:
@amboar I built an image with this and net booted and still seeing 255 for all LEDs off 955x

Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Thu Aug 24 15:10:15 IST 2017 armv6l GNU/Linux

root@witherspoon:/sys/class/leds# cat rear-power/brightness 
255

root@witherspoon:/sys/class/leds# cat front-id/brightness 
255

root@witherspoon:/sys/class/leds# cat fan0//brightness 
255

```
diff --git a/meta-phosphor/common/recipes-kernel/linux/linux-obmc.inc b/meta-phosphor/common/recipes-kernel/linux/linux-obmc.inc
index 72819bf..b66c328 100644
--- a/meta-phosphor/common/recipes-kernel/linux/linux-obmc.inc
+++ b/meta-phosphor/common/recipes-kernel/linux/linux-obmc.inc
@@ -12,6 +12,7 @@ SRC_URI += "file://phosphor-gpio-keys.scc"
 SRC_URI += "file://phosphor-gpio-keys.cfg"
 SRC_URI += "file://phosphor-vlan.scc"
 SRC_URI += "file://phosphor-vlan.cfg"
+SRC_URI += "file://0001-leds-pca955x-Don-t-invert-requested-value-in-pca955x.patch"
```

```
Untracked files:
  (use "git add <file>..." to include in what will be committed)

        meta-phosphor/common/recipes-kernel/linux/linux-obmc/0001-leds-pca955x-Don-t-invert-requested-value-in-pca955x.patch

```

vishwabmc, 2017-08-24:
Not sure what went wrong in the test I did before.. I went ahead tested 2 more times and saw that it worked good net booting.
```
root@witherspoon:/sys/class/leds# cat fan0/brightness fan1/brightness fan2/brightness fan3/brightness 
0
0
0
0
```
```
root@witherspoon:/sys/class/leds# cat front-fault/brightness front-id/brightness front-power/brightness rear-fault/brightness rear-id/brightness rear-power/brightness 
0
0
0
0
0
0
```
```
root@witherspoon:/sys/class/leds# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Thu Aug 24 15:10:15 IST 2017 armv6l GNU/Linux
root@witherspoon:/sys/class/leds# 
```

amboar, 2017-08-24:
Well I'm mildly surprised at that outcome; didn't think I'd resolve it first go. I actually haven't tried the patch myself (due to lack of machines to test on amongst other distractions), but if I get some time tomorrow I'll try to reproduce your result

amboar, 2017-08-25:
@vishwabmc I've reproduced both of your results above. The rebooting from the broken kernel to the 'fixed' kernel they come back full brightness, but subsequent reboots of the 'fixed' kernel see them retain their 0 value.

However, with the 'fixed' kernel, now setting the leds to full brightness then rebooting into the same kernel leads the values to returning to 0.

I think something is funky in the remove() path for the driver.

amboar, 2017-08-25:
o.O

```c

static void gpio_led_shutdown(struct platform_device *pdev)
{
	struct gpio_leds_priv *priv = platform_get_drvdata(pdev);
	int i;

	for (i = 0; i < priv->num_leds; i++) {
		struct gpio_led_data *led = &priv->leds[i];

		gpio_led_set(&led->cdev, LED_OFF);
	}
}

static struct platform_driver gpio_led_driver = {
	.probe		= gpio_led_probe,
	.shutdown	= gpio_led_shutdown,
	.driver		= {
		.name	= "leds-gpio",
		.of_match_table = of_gpio_leds_match,
	},
};
```

amboar, 2017-08-25:
I've sent out a series fixing the issue: https://lists.ozlabs.org/pipermail/openbmc/2017-August/008923.html

@bjwyman @vishwabmc @geissonator @legoater can you please review/test the patches?

bjwyman, 2017-08-25:
Pulled down mbox patches:
http://patchwork.ozlabs.org/patch/805788/
http://patchwork.ozlabs.org/patch/805789/
http://patchwork.ozlabs.org/patch/805790/
http://patchwork.ozlabs.org/patch/805791/
http://patchwork.ozlabs.org/patch/805792/
Copied to openbmc/meta-phosphor/common/recipes-phosphor/linux/linux-obmc/
Updated linux-obmc.inc to add SRC_URI pointing to those files.
(also had local changes for issue 1732)
Built image. Netbooted on pass-1 Sequoia system in Rochester.
```root@witherspoon:/sys/class/leds# grep . */brightness
fan0/brightness:0
fan1/brightness:0
fan2/brightness:0
fan3/brightness:0
front-fault/brightness:0
front-id/brightness:0
front-power/brightness:0
rear-fault/brightness:0
rear-id/brightness:0
rear-power/brightness:0
root@witherspoon:/sys/class/leds#
```
The front-power and rear-power are occasionally 255 (due to slow blink at power off state).
Went to the lab, verified front LEDs looked to be in the correct state. All four amber fan LEDs off, green power LED in slow blink (1 Hz?), amber and blue LED off.
I am not certain on the saving of the state though. Since this is a pass-1 system with DD1 processors, I cannot power it on and try the reset/reload.
I might be able to try a DD2 system, but would likely need to flash the image onto such a system and get some remote help to check LED states.

amboar, 2017-08-25:
Thanks @bjwyman. Do you mind sending Tested-by tags for each of the patches to the list?

bjwyman, 2017-08-28:
Okay, I think I have the "Tested-by" tag in each of the patches.

rfrandse, 2017-08-30:
Fixed

closed 2017-08-30 by rfrandse

bjwyman, 2017-08-30:
https://gerrit.openbmc-project.xyz/#/c/6271/
https://github.com/openbmc/linux/commit/70e22d54acb839ba89da4afeda04ddd8c77300a3
https://github.com/openbmc/linux/commit/4c148ccf851cb28ab95958c9ca4e4beb54559fa1
https://github.com/openbmc/linux/commit/f1d699831b14119c703cd0be06ea10b48413692c
https://github.com/openbmc/linux/commit/79485a11258a2bbf360e88f89ab9118b04728d32
https://github.com/openbmc/linux/commit/20d7f38141d789a89ae78a8e8245882d56408dff
https://github.com/openbmc/linux/commit/76ff93cf4565068e97b802e41748eb6745500606