openbmc/openbmc #2157: obmc-flash-bios: error 32 - mount: wrong fs type, bad option, bad superblock on ubi7:pnor-prsv

gkeishin, 2017-08-16:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-256"
VERSION_ID="v1.99.8-256-gb25ba23"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-256"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Wed Aug 16 04:20:42 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
	"/xyz/openbmc_project/software/7c50f790": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Ready",
        "ExtendedVersion": "op-build-v1.18-62-gbf34a31-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-76-g830fc9a0ed0a-pa97b453,hostboot-42ecd45,linux-4.12.5-openpower1-p3d0b449,petitboot-v1.5.1-p5fba69a,machine-xml-4b12db6-p25651a7,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "/tmp/images/7c50f790",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.18_1.25"
    },

	
   "/xyz/openbmc_project/software/7c50f790": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Failed",
        "ExtendedVersion": "op-build-v1.18-62-gbf34a31-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-76-g830fc9a0ed0a-pa97b453,hostboot-42ecd45,linux-4.12.5-openpower1-p3d0b449,petitboot-v1.5.1-p5fba69a,machine-xml-4b12db6-p25651a7,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "/tmp/images/7c50f790",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active",
        "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.18_1.25"
    },
```
`Journald snippet`
```

Aug 03 15:48:48 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbios\x2dsquashfsmount.slice.
Aug 03 15:48:48 witherspoon systemd[1]: Starting Mount SquashFS image pnor-ro-7c50f790...
Aug 03 15:48:49 witherspoon systemd[1]: Created slice system-obmc\x2dflash\x2dbios\x2dubimount.slice.
Aug 03 15:48:49 witherspoon systemd[1]: Starting Mount UBIFS volume pnor-prsv and pnor-rw-7c50f790...
Aug 03 15:48:49 witherspoon phosphor-gevent[1286]: ::80f9:6076:8e91:f402%1995176732 - - [2017-08-03 15:48:49] "PUT /xyz/openbmc_project/software/7c50f790/attr/RequestedActivation HTTP/1.1" 200 169 0.487970

Aug 03 15:49:11 witherspoon obmc-flash-bios[1391]: Volume ID 0, size 257 LEBs (16809856 bytes, 16.0 MiB), LEB size 65408 bytes (63.9 KiB), dynamic, name "pnor-rw-7c50f790", alignment 1
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): default file-system created
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): UBIFS: mounted UBI device 7, volume 0, name "pnor-rw-7c50f790"
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): reserved for root: 763076 bytes (745 KiB)
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): media format: w5/r0 (latest is w5/r0), UUID C27B7C2D-52F5-4E3B-98DC-02F772694349, small LPT model
Aug 03 15:49:11 witherspoon kernel: UBIFS (ubi7:0): background thread "ubifs_bgt7_0" started, PID 1419
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): default file-system created
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): UBIFS: mounted UBI device 7, volume 0, name "pnor-rw-7c50f790"
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): FS size: 16155776 bytes (15 MiB, 247 LEBs), journal size 784896 bytes (0 MiB, 12 LEBs)
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): reserved for root: 763076 bytes (745 KiB)
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): media format: w5/r0 (latest is w5/r0), UUID C27B7C2D-52F5-4E3B-98DC-02F772694349, small LPT model
Aug 03 15:49:11 witherspoon kernel[979]: [  385.520000] UBIFS (ubi7:0): background thread "ubifs_bgt7_0" started, PID 1419
Aug 03 15:49:12 witherspoon obmc-flash-bios[1420]: mount: wrong fs type, bad option, bad superblock on ubi7:pnor-prsv,
Aug 03 15:49:12 witherspoon obmc-flash-bios[1420]:        missing codepage or helper program, or other error
Aug 03 15:49:12 witherspoon obmc-flash-bios[1420]:        In some cases useful info is found in syslog - try
Aug 03 15:49:12 witherspoon obmc-flash-bios[1420]:        dmesg | tail or so.
Aug 03 15:49:12 witherspoon obmc-flash-bios[1420]: /usr/sbin/obmc-flash-bios: error 32
Aug 03 15:49:12 witherspoon systemd[1]: obmc-flash-bios-ubimount@7c50f790.service: Main process exited, code=exited, status=32/n/a
Aug 03 15:49:12 witherspoon systemd[1]: Failed to start Mount UBIFS volume pnor-prsv and pnor-rw-7c50f790.
Aug 03 15:49:12 witherspoon systemd[1]: obmc-flash-bios-ubimount@7c50f790.service: Unit entered failed state.
Aug 03 15:49:12 witherspoon systemd[1]: obmc-flash-bios-ubimount@7c50f790.service: Failed with result 'exit-code'.
Aug 03 15:49:12 witherspoon kernel: UBIFS error (ubi7:2 pid 1429): ubifs_recover_master_node: failed to recover master node
Aug 03 15:49:12 witherspoon kernel[979]: [  386.790000] UBIFS error (ubi7:2 pid 1429): ubifs_recover_master_node: failed to recover master node

Aug 03 15:50:05 witherspoon obmc-flash-bios[1390]: Volume ID 1, size 385 LEBs (25182080 bytes, 24.0 MiB), LEB size 65408 bytes (63.9 KiB), static, name "pnor-ro-7c50f790", alignment 1
Aug 03 15:50:05 witherspoon obmc-flash-bios[1390]: ubiupdatevol: error!: stat failed on "/tmp/images/7c50f790/pnor.xz.squashfs"
Aug 03 15:50:05 witherspoon obmc-flash-bios[1390]: mount: /dev/ubiblock7_1: can't read superblock
Aug 03 15:50:05 witherspoon obmc-flash-bios[1390]: /usr/sbin/obmc-flash-bios: error 32
Aug 03 15:50:05 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@7c50f790.service: Main process exited, code=exited, status=32/n/a
Aug 03 15:50:05 witherspoon systemd[1]: Failed to start Mount SquashFS image pnor-ro-7c50f790.
Aug 03 15:50:05 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@7c50f790.service: Unit entered failed state.
Aug 03 15:50:05 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@7c50f790.service: Failed with result 'exit-code'.
Aug 03 15:50:06 witherspoon kernel: block ubiblock7_1: created from ubi7:1(pnor-ro-7c50f790)
Aug 03 15:50:06 witherspoon kernel: blk_update_request: I/O error, dev ubiblock7_1, sector 0
Aug 03 15:50:06 witherspoon kernel: SQUASHFS error: squashfs_read_data failed to read block 0x0
Aug 03 15:50:06 witherspoon kernel: squashfs: SQUASHFS error: unable to read squashfs_super_block
Aug 03 15:50:06 witherspoon kernel[979]: [  439.700000] block ubiblock7_1: created from ubi7:1(pnor-ro-7c50f790)
Aug 03 15:50:06 witherspoon kernel[979]: [  439.700000] blk_update_request: I/O error, dev ubiblock7_1, sector 0
Aug 03 15:50:06 witherspoon kernel[979]: [  439.700000] SQUASHFS error: squashfs_read_data failed to read block 0x0
Aug 03 15:50:06 witherspoon kernel[979]: [  439.700000] squashfs: SQUASHFS error: unable to read squashfs_super_block
```

- [x] Journald log 
[journald_v1.99.8-256-gb25ba23_PNOR_update.txt](https://github.com/openbmc/openbmc/files/1227192/journald_v1.99.8-256-gb25ba23_PNOR_update.txt)

geissonator, 2017-08-16:
#2109 maybe?

gkeishin, 2017-08-16:
That looks similar signature from the journald. @saqibkh can take a stab at the log here.  just in case needed to see the latest failure. by the way, this is blocking the Master test.  

I ll close this one up once he confirms it.

mdmillerii, 2017-08-17:
The ubiblock stuff definitely.   Not sure about the bad superblock on `ubi7:pnor-prsv` but it could be related to the history of the partition.

saqibkh, 2017-08-22:
We are hoping that #2109 would have this bug.
@gkeishin can you confirm that the issue is resolved.
Thanks

gkeishin, 2017-08-22:
I ll watch out in next Master test and close this one if not seen.. Thanks

gkeishin, 2017-08-22:
This is fix and @charleshofer also didn't see in latest test.. closing up..

closed 2017-08-22 by gkeishin