openbmc/openbmc #2161: openpower-occ-control application fails on re-ipl

geissonator, 2017-08-16:
The first boot, everything worked great and both occ's hwmon's started correctly.  On the second IPL, we hit this:

```
Aug 03 18:25:07 witherspoon ipmid[1190]: Bus: :1.64, Path: /org/open_power/control/occ1, Interface: org.open_power.OCC.Status
Aug 03 18:25:07 witherspoon ipmid[1190]: ERROR:[0xCB] handling NetFn:[0x4], Cmd:[0x2D]
Aug 03 18:25:09 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Aug 03 18:25:10 witherspoon openpower-occ-control[1150]: Opening OCC device failed.
Aug 03 18:25:10 witherspoon openpower-occ-control[1150]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::OpenFailure'
Aug 03 18:25:10 witherspoon openpower-occ-control[1150]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Aug 03 18:25:10 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Aug 03 18:25:10 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Aug 03 18:25:10 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Aug 03 18:25:11 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
```

The callout info in the journal:
```
        "CALLOUT_ERRNO" : "2",
        "CALLOUT_DEVICE_PATH" : "/dev/occ2",
```

FYI @vishwabmc

geissonator, 2017-08-16:
I think the issue is this code:
```
22            if (!hubFsiScanDone)
23            {
24                // Need to do hub scan before we bind
25                this->scanHubFSI();
26            }
```

We need to do the hubFsiScan on every boot, not just the first.

rfrandse, 2017-08-16:
<https://gerrit.openbmc-project.xyz/6039> Reset hub scan variable when OCC goes inactive
Resolves  #2161 openpower-occ-control application fails on re-ipl

closed 2017-08-17 by williamspatrick