openbmc/openbmc #2163: [xCAT] Using the REST commands for Network, Properties do not take...

whowutwut, 2017-08-17:
@ratagupt I got around to playing with this today...   some very strange issues arise..   

1.  I set the IP address using busctl to 9.114.120.110  .  I don't have a DHCP server for this node , so I can't set it to an open range.   I know you said we should not use busctl. 

2. Script that does the following curl commands, set to .111 IP address... 
```
[root@stratton01 frame6u05]# ./set_ip.sh
++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": "9.114.120.111" }' https://9.114.120.110//xyz/openbmc_project/network/eth0/ipv4/724c9cc6/attr/Address
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": 24 }' https://9.114.120.110//xyz/openbmc_project/network/eth0/ipv4/724c9cc6/attr/PrefixLength
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": "frame6u05-bmc" }' https://9.114.120.110//xyz/openbmc_project/network/config/attr/HostName
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

3. Reboot the BMC (How do we activate the settings without rebooting???)

4. When the BMC comes back up... the hostname is changed, but the IP address is still 9.114.120.110.... via ifconfig. 

5. Ran the above script again, list out... to prove the settings... 

Before:
```
[root@stratton01 frame6u05]# ./list.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://9.114.120.110/xyz/openbmc_project/network/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/config": {
      "DefaultGateway": "9.114.120.1",
      "HostName": "frame6u05-bmc"
    },
   ...
    "/xyz/openbmc_project/network/eth0/ipv4/724c9cc6": {
      "Address": "9.114.120.110",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 24,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
   ...
  "message": "200 OK",
  "status": "ok"
}
```

Set it again
```
[root@stratton01 frame6u05]# ./set_ip.sh
++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": "9.114.120.111" }' https://9.114.120.110//xyz/openbmc_project/network/eth0/ipv4/724c9cc6/attr/Address
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": 24 }' https://9.114.120.110//xyz/openbmc_project/network/eth0/ipv4/724c9cc6/attr/PrefixLength
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ '[' '!' -z x ']'
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data": "frame6u05-bmc" }' https://9.114.120.110//xyz/openbmc_project/network/config/attr/HostName
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

AFter reboot of BMC... and logging into the BMC, something happened where the BMC switched itself back to an Old version of the BMC...  ...   @geissonator is aware of this behavior in the last 1-2 days on this node... ...
```
root@witherspoon:~# cat /etc/*release*
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.4"
VERSION_ID="ibm-v1.99.4-0-r0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.4"
BUILD_ID="ibm-v1.99.4-0-r0"
```

and the IP address is old... 
```
root@witherspoon:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.254  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::72e2:84ff:fe14:280d%1334260624/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:277 errors:0 dropped:0 overruns:0 frame:0
          TX packets:367 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:13930 (13.6 KiB)  TX bytes:27922 (27.2 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1334260624/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1600 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1600 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:121600 (118.7 KiB)  TX bytes:121600 (118.7 KiB)

root@witherspoon:~# ```

Besides the revert to an OLD BMC level...  Am i calling this correctly? 
Is there a way to activate the network settings without rebooting the BMC?

whowutwut, 2017-08-17:
I talked to @ratagupt and @williamspatrick  and it seems like this is not supported.  The ability to change existing network object settings.   If that is correct, then I can close this issue?  Will this be supported in the future? 

The recommended method is to create a new network object and then delete the old one...

ratagupt, 2017-08-24:
@whowutwut :- IPMI will work as it was done earlier,
Earlier we were not supporting IP aliasing through REST,which we have started supporting.
Modification is not supported now through REST,But still you can achieve the same with delete/add.
We can support the modification of the existing ip object.

whowutwut, 2017-08-27:
@ratagupt can you clarify the last sentence?

Do you mean right now you don't support modification but can in the future?

ratagupt, 2017-08-31:
@vhu:- I will take it as enhnacement,and I don't consider it for phase 5.

whowutwut, 2017-08-31:
What is the date for Phase 5? where can we see that schedule?

ratagupt, 2017-09-04:
@whowutwut 
Phase 5:- 09/06
Phase5.1:- 09/29
We will take it as phase 5.1..Till then you can achieve the same functionality by
1) Delete existing ip object
2) Add the IP object with the desired parameters.

gtmills, 2018-07-05:
https://github.com/openbmc/openbmc/issues/3227 is a similiar issue.

gtmills, 2018-09-26:
We still have a hack in the GUI for this (have to remove then add an IPV4 interface). Really would be nice to modify existing Network IPV4 interfaces.

closed 2019-04-27 by stale[bot]