openbmc/openbmc #2164: HOST is not powering on after updating the latest GHE op910.1734.20170812q PNOR build

PavamanSubramaniyam, 2017-08-17:
I have installed the P9 witherspoon Hardware with latest master open bmc firmware image from the link:
https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/979/artifact/deploy/images/witherspoon/witherspoon-20170816101614.all.tar

Also, tried updating the PNOR Firmware image from the latest GHE build available in the link:
https://w3.austin.ibm.com/afs/austin/projects/esw/oppnor910/Builds/op910.1734.20170812q/op-build/output/images/ship/witherspoon-sequoia.pnor.squashfs.tar

After these firmware updates, when we try to do obmcutil poweron, the machine is not powering on the host.

``
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

PavamanSubramaniyam, 2017-08-17:
```
# curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://<BMC IP>/login
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}

# curl -c cjar -b cjar -k -H "Content-Type: application/json" https://<BMC IP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/81fc9494": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-258-gc9a4174"
    },
    "/xyz/openbmc_project/software/f4a128ac": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.18-68-gd63eb15-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-42ecd45,linux-4.12.5-openpower1-p795b922,petitboot-v1.5.1-p66ffa0d,machine-xml-9a28007-p280c38b,occ-71b5f68-p8e32a5a,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.30"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

PavamanSubramaniyam, 2017-08-17:
[20170817100015919833_Myffdc.zip](https://github.com/openbmc/openbmc/files/1230894/20170817100015919833_Myffdc.zip)

geissonator, 2017-08-22:
@PavamanSubramaniyam - you still seeing issues here?  This should all be in op-build now from @charleshofer so please ping him if you have any questions on the build of the .squashfs.tar image.  Any general flash and boot questions, please ping @gkeishin.

closed 2017-08-22 by geissonator