openbmc/openbmc #2166: IPL: core_phosphor-hwmon on w55 (DD20) with 910.1734.20170812c

cdeadmin, 2017-08-17:


spinler, 2017-08-17:
A phosphor-hwmon instance for a VRM is dying on a power on:

```

Aug 16 17:33:52 witherspoon phosphor-hwmon-readd[1593]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Aug 16 17:33:52 witherspoon phosphor-hwmon-readd[1593]:   what():  filesystem error: cannot canonicalize: No such file or directory [/sys/class/hwmon/hwmon5/of_node] [/]
Aug 16 17:33:53 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Aug 16 17:33:53 witherspoon kernel[938]: [  482.500000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Aug 16 17:33:53 witherspoon systemd[1]: Starting Assert power_on LED...
```
Unbinding the driver is normal during a power on, though normally if the driver is unbound in the middle of an operation, the read will fail with a ENOENT and we'll handle that gracefully.  Here it appears we got a different errno so we tried to do a callout and then crashed because the driver was gone.

Could fix by either:
* treat a missing callout path like an ENOENT and exit(0)
* do a callout without the fs::canonical if the path doesn't exist

bradbishop, 2017-08-21:
I had taken an initial stab at this here: https://gerrit.openbmc-project.xyz/#/c/5071/

rfrandse, 2017-08-30:
<https://gerrit.openbmc-project.xyz/6303> Use hwmonio for attribute IO
Resolves  #2166 IPL: core_phosphor-hwmon on w55 (DD20) with 910.1734.20170812c

closed 2017-09-05 by williamspatrick