openbmc/openbmc #2167: Sprint 17.08.23 - Committed

rfrandse, 2017-08-17:


dkodihal, 2017-08-21:
***Squad 1 Sprint Plan***

**Vishwa**
#1714 - REST API to set a user login password

**Deepak**
#2102 - External Event Notification : design and prototype
#1856 Design event logging mechanism

**Tom**
#1957 - DCMI: Design app for data sampling/telemetry.
Defects
#1677 - Boot count not getting correctly read and updated by hostboot

**Jayanth**
#1623 - Enable application core dump file
New dump story and/or #2078 may carry over.

**Nagaraju**
#1561 - Request for Delete "all" Event Log command

**Ratan**

**Dhruv**

**Devender**

spinler, 2017-08-23:
# Squad 2 -

 **Andrew:geissonator -**

**17.08.09:**

- #2148 (unset occ active on power down)

- #2129 (SSH intermittently stops working)

- #2162 (boot shutdown timeout increase)

- #2161 (occ control re-ipl bug fix)

- #2148 (unset occ active on power down)

**17.08.23:**

---

**Chris:cbostic:**

**17.07.12:**

- #1727 (ucd driver support)
  * 

**17.08.09:**

- #2191 (cfam on 2nd proc bug)
  * 08/22 - Have workaround, still digging into final solution

- #1974 (support for booting off alternate bmc flash chip)

**17.08.23:**

---

**Brandon:bjwyman:**

**17.07.26:**

- #1731 (power supply monitor for faults)
  * 

**17.08.23:**

- #1732 (power faults at runtime)
  * 

---

**Eddie:eddiejames:**

**17.08.23:**

---

**Matt S:spinler:**

**17.06.28:**
-  #1724 (UCD chip fault analysis) 

**17.08.23:**

- #1726 (Analyze GPU PGOOD fault or OverTemp down to the GPU)

---

**Matt B:msbarth:**

**17.08.09:**

- **#1792** (yaml updates)

- #2187 (need to reset functional fans on boot)

- #1922 (fan override ability)

- #2159 (fan override)

---

**Brad:bradbishop:**

**17.08.23:**

---

**Diniesh:**

**17.08.23:**

anoo1, 2017-08-23:
# Squad 3 -

**Adriana:anoo1 -**
**17.08.23**
- #1967 Setting boot policy causes system to boot from alternate bmc chip [Witherspoon]
- #2200 BMC Update: Write kernel to both chips and update boot env only on success
- #2201 BMC Update: Update documentation
---
**Gunnar:gtmills -**
**17.08.23**
- #2007 ItemUpdater: Implement Associations
- #1898 [security] overwrite any file on the bmc with updateViaTftp
- #2190 Create error log with message field on system shutdown due to 3 cores over 115C
---
**Eddie:eddiejames -**
**17.08.23**
- #2086 [xCAT] Takes 2 delete actions to remove PNOR image in Ready state.
- #1986 Prevent Delete on currently running version
---
**Saqib:saqibkh -**
**17.08.23**
- #2199 BMC: Mount /usr/local/ in read-only filesystem for patching
- #2172 rename '/dev/mtd/.tmp-c90:3' '/dev/mtd/' failed: Not a directory
- #2158 systemd-journal: /dev/kmsg buffer overrun
---
**Michael:mtritz -**
**17.08.23**
- #2145 UBIFS error (pid: 1033): cannot open "ubi7:pnor-patch", error -19 is seen in dmesg
- #1721 Factory reset: Define a generic ResetFailed error
- #1466 discover-system-state: Accept a node parameter
---
**Leo:lgon -**
**17.08.23**
- #1663 Rename mboxd dbus object to be in correct xyz.openbmc_project
- #2072 Make /media tmpfs
- #1539 Update rest_dbus.py MethodHandler Class to Convert Method Arguments
---
**Charlie: -** 

---

closed 2017-09-14 by zahrens