openbmc/openbmc #2170: openbmc kernel boot fail on ast2500 customized board

LindarLiu0724, 2017-08-18:
Hi all,
I use latest openbmc and meet below kernel crash, could somebody give some suggestions for my debug.

thanks very much
--------------------------------log---------------------------------------------
done
Bytes transferred = 33554432 (2000000 hex)
AST2500EVB>
DRAM Init-V10-DDR4
0abc1-4Gb-Done
Read margin-DL:0.3411/DH:0.3470 CK (min:0.30)

U-Boot 2016.07 (Aug 16 2017 - 09:55:13 +0800)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC0 : RGMII
MAC1 : RGMII
FTGMAC100#0
Error: FTGMAC100#0 address not set.
, FTGMAC100#1
Error: FTGMAC100#1 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x2008012c
     Data Size:    2078208 Bytes = 2 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   78443adbd21bd884ac6dca015458be2e750ae831
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x2028117c
     Data Size:    1585383 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   9f458b6316320cce31259ec02ec4ddfac09aac9b
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x2027b820
     Data Size:    22683 Bytes = 22.2 KiB
     Architecture: ARM
     Hash algo:    sha1
[... 134 lines cut ...]
[    2.810000] 1f40: 9e4a1f94 9e4a1f50 8058b040 8058add0 00000006 00000006 00000000 8058a6b0
[    2.810000] 1f60: 3940bf3d 00000007 f115c2e8 00000000 803fd158 00000000 00000000 00000000
[    2.810000] 1f80: 00000000 00000000 9e4a1fac 9e4a1f98 803fd170 8058af38 00000000 803fd158
[    2.810000] 1fa0: 00000000 9e4a1fb0 8000a5e8 803fd164 00000000 00000000 00000000 00000000
[    2.810000] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    2.810000] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 d11dcae8 af8ddec5
[    2.810000] [<80331004>] (netif_napi_del) from [<80292b30>] (ftgmac100_probe+0x290/0x674)
[    2.810000] [<80292b30>] (ftgmac100_probe) from [<80243df8>] (platform_drv_probe+0x60/0xc0)
[    2.810000] [<80243df8>] (platform_drv_probe) from [<80242178>] (driver_probe_device+0x280/0x44c)
[    2.810000] [<80242178>] (driver_probe_device) from [<80242434>] (__driver_attach+0xf0/0x104)
[    2.810000] [<80242434>] (__driver_attach) from [<80240148>] (bus_for_each_dev+0x7c/0xb0)
[    2.810000] [<80240148>] (bus_for_each_dev) from [<802425dc>] (driver_attach+0x28/0x30)
[    2.810000] [<802425dc>] (driver_attach) from [<80240ba8>] (bus_add_driver+0x14c/0x268)
[    2.810000] [<80240ba8>] (bus_add_driver) from [<80243068>] (driver_register+0x88/0x104)
[    2.810000] [<80243068>] (driver_register) from [<80244a38>] (__platform_driver_register+0x40/0x54)
[    2.810000] [<80244a38>] (__platform_driver_register) from [<805ab264>] (ftgmac100_driver_init+0x18/0x20)
[    2.810000] [<805ab264>] (ftgmac100_driver_init) from [<8058ae70>] (do_one_initcall+0xac/0x168)
[    2.810000] [<8058ae70>] (do_one_initcall) from [<8058b040>] (kernel_init_freeable+0x114/0x1cc)
[    2.810000] [<8058b040>] (kernel_init_freeable) from [<803fd170>] (kernel_init+0x18/0x104)
[    2.810000] [<803fd170>] (kernel_init) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[    2.810000] Code: e594205c e5941058 e2843058 e3a05000 (e5812004) 
[    3.210000] ---[ end trace f32811052fd3860c ]---
[    3.210000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    3.210000] 
[    3.210000] Reboot failed -- System halted

amboar, 2017-08-21:
Hi @LindarLiu0724, first up - a note about formatting issues in github:

Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```
Onto the panic, you mention in the issue subject an 'ast2500 customised board'. Does the board have a dedicated network PHY or NC-SI? The devicetree you've booted the kernel with is likely incorrect for your hardware setup.

LindarLiu0724, 2017-08-21:
Thanks for your guide.
Yes, we use a dedicated network PHY BCM54612E, is any example for this phy configuration?

amboar, 2017-08-21:
@LindarLiu0724 From a quick survey of our dts files we don't appear to specify the PHY, but simply add or remove the `use-ncsi` property in the `macX` nodes as appropriate.

Did you use one of our in-tree devicetree files and if so, which one? Alternatively have you written your own? If you have can you provide a link to it?

LindarLiu0724, 2017-08-21:
We now use in-tree devicetree aspeed-ast2500-evb.dts.

amboar, 2017-08-21:
> We now use in-tree devicetree aspeed-ast2500-evb.dts.

This is what you were using when you created the issue? Or have you changed the devicetree you were using since opening it?

LindarLiu0724, 2017-08-21:
Yes.
We haven't make any change.

amboar, 2017-08-21:
@LindarLiu0724 so I don't know why your PHY isn't found by the MDIO search, however we shouldn't be crashing regardless. Please test my fix at https://github.com/amboar/linux/commit/94320148da7a6d647ca40dc63ade465c01c0f07c and let me know if that helps. If so I'll send it to the relevant lists.

A quick test in qemu suggests it should stop the kernel from crashing.

LindarLiu0724, 2017-08-21:
Do you mean that use in-tree devicetree aspeed-ast2500-evb.dts should work?
I will test you fix and give you feedback.
thanks.

amboar, 2017-08-22:
> Do you mean that use in-tree devicetree aspeed-ast2500-evb.dts should work?

Without knowledge of your hardware design it's hard to tell.

The patch fixes a separate problem (the crash in the MAC driver *after* we fail to find the PHY). By including it you should at least be able to boot to userspace, though networking won't be functional.

LindarLiu0724, 2017-08-22:
The patch works well, no OOP now. seems ethernet also alive.
We can see the login. could you tell me the login user and password?
thanks very much.

amboar, 2017-08-22:
> The patch works well, no OOP now. 

Great!

> seems ethernet also alive.

Maybe the failure was on the second MAC? Looking back at the logs the first MAC didn't fail to find the PHY.

> We can see the login. could you tell me the login user and password?

The defaults are:

user: root
pass: 0penBmc

Note that 0penBmc is spelled with the digit character 'zero', not the letter 'oh', so &lt;zero&gt;penBmc.

LindarLiu0724, 2017-08-22:
yes, we just use one MAC, now ethertnet works ok.
We can login now. :)
And what's the user/password for ipmi?

Another questions need your kindly help: 
Is any document to introduce how to update flash from u-boot command line?
now we use AMI u-boot to flash opembmc image, and it works.
but if we use openbmc uboot to flash AMI image and it fail.

amboar, 2017-08-22:
> yes, we just use one MAC, now ethertnet works ok.

Okay, you should probably change your devicetree to set

```
&mac1 {
    status = "disabled";
};

By doing that you would avoid the second mac being probed, which in turn would have worked around the bug. On another note I've sent the patch upstream, you can track it here:

https://patchwork.ozlabs.org/patch/804288/

> We can login now. :)

Excellent.

> And what's the user/password for ipmi?
>
> Another questions need your kindly help:
> Is any document to introduce how to update flash from u-boot command line?
> now we use AMI u-boot to flash opembmc image, and it works.
> but if we use openbmc uboot to flash AMI image and it fail.

I suggest you post your questions to the list. That way they get more visibility and other people are able to help out. I'd like to keep this issue focused on the network bug, as with your confirmation that the patch worked for you we can close it.

LindarLiu0724, 2017-08-22:
OK, thanks for your kindly help.  Let's close it.

amboar, 2017-08-22:
Well, I spoke to soon, we should wait for feedback and get @shenki to apply it to openbmc/linux, then close it, but that's hopefully not far off :)

amboar, 2017-08-23:
@LindarLiu0724 upstream have applied the patch, and I've poked @shenki to backport it to openbmc/linux dev-4.10. Should be appearing in the tree in the near future.

LindarLiu0724, 2017-08-23:
Good, thanks for updating.

amboar, 2017-08-23:
@LindarLiu0724 The patch has been applied as openbmc/linux@6246810861383c08c928c1df651826d931284f50. If you're using openbmc/linux as a remote for your kernel tree you can now fetch it in. openbmc/openbmc also has a kernel bump request open which pulls the change into yocto:

https://gerrit.openbmc-project.xyz/#/c/6201/

closed 2017-08-23 by williamspatrick