openbmc/openbmc #2171: Network:- Implement the Network Monitor app

ratagupt, 2017-08-18:
Impelement the network monitor app which will listen on the new address event ,which sends the dbus message to the network manager app to refresh its interfaces to reflect the action on the interface.

ratagupt, 2017-08-24:
- [ ] Create an app which listens on systemd signal (if systemd-networkd sends it) / RTNETlink addADDR signal on an interface 

- [ ] After getting signal on a particular interface it will ask the networkmanager to refresh the IP objects of the given interface.

ratagupt, 2017-09-01:
# **STATUS**
08/31
Discussed the approach in the design meeting.
01/09
Haven't started the coding.

rfrandse, 2017-09-04:
<https://gerrit.openbmc-project.xyz/6399> Network: Add network Monitor
Resolves  #2171 Network:- Implement the Network Monitor app

closed 2017-09-13 by williamspatrick