openbmc/openbmc #2172: rename '/dev/mtd/.tmp-c90:3' '/dev/mtd/' failed: Not a directory

gkeishin, 2017-08-19:
```
Aug 03 15:43:02 witherspoon systemd-udevd[694]: rename '/dev/mtd/.tmp-c90:3' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:02 witherspoon systemd-udevd[699]: Could not generate persistent MAC address for eth0: No such file or directory
Aug 03 15:43:03 witherspoon systemd-udevd[698]: rename '/dev/mtd/.tmp-c90:1' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:03 witherspoon systemd-udevd[701]: rename '/dev/mtd/.tmp-c90:13' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:03 witherspoon systemd-udevd[695]: rename '/dev/mtd/.tmp-c90:15' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:03 witherspoon systemd-udevd[700]: rename '/dev/mtd/.tmp-c90:11' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:03 witherspoon systemd-udevd[702]: rename '/dev/mtd/.tmp-c90:9' '/dev/mtd/' failed: Not a directory
Aug 03 15:43:03 witherspoon systemd-udevd[694]: rename '/dev/mtd/.tmp-c90:5' '/dev/mtd/' failed: Not a directory
```
```

root@witherspoon:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     215208         0    215208   0% /dev
tmpfs                   216124     14348    201776   7% /run
/dev/mtdblock4           14464     14464         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       308      3788   8% /run/initramfs/rw
cow                       4096       308      3788   8% /
tmpfs                   216124         0    216124   0% /dev/shm
tmpfs                   216124         0    216124   0% /sys/fs/cgroup
tmpfs                   216124         0    216124   0% /tmp
tmpfs                   216124         0    216124   0% /usr/local
tmpfs                   216124        84    216040   0% /var/volatile
/dev/ubiblock7_0         19712     19712         0 100% /media/pnor-ro-fd2cbe15
ubi7:pnor-rw-fd2cbe15
                         13816        16     13052   0% /media/pnor-rw-fd2cbe15
ubi7:pnor-prsv             972        32       872   4% /media/pnor-prsv
ubi7:pnor-patch          13816        16     13056   0% /usr/local/share/pnor
root@witherspoon:~#
```

`Driver Info`
```
root@witherspoon:~# cat /etc/timestamp
20170819012228
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Sat Aug 19 01:21:10 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

$ ./rest_cmd -i wsbmc006 -o GET -u /xyz/openbmc_project/software/enumerate
Executing: GET  /xyz/openbmc_project/software/enumerate

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/686a9761": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-292-g92b1559"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
}
```
[journald_v1.99.8-292-g92b1559_Update.txt](https://github.com/openbmc/openbmc/files/1236197/journald_v1.99.8-292-g92b1559_Update.txt)

saqibkh, 2017-08-24:
@williamspatrick I wanted to run this by you first before opening a defect to systemd.

^ its a pretty early journald log and comes up when the system is booting up.
I'm not seeing anything helpful on the metadata of the journal and also couldn't find anything similar online.

mdmillerii, 2017-08-24:
Why is udev trying to rename to `/dev/mtd` ?
Sounds suspiciously like our `/dev/mtd/<name>` rule is not mtd names.

The mtd `name` attribute is via the `device_type` `groups` hook and should not be racy, but there are 7 errors and 7 mtds.

It it an issue creating the `/dev/mtd` directory ?  @eddiejames

eddiejames, 2017-08-29:
# **STATUS**

08/29 - Not sure why it would be trying to rename stuff in /dev/mtd/. The directory should be created by udev. Maybe a formatting issue with my udev rule? I don't see this on all systems, in fact I haven't recreated yet.

gkeishin, 2017-09-07:
It still persists in Master and we can see on boot to standby.

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6511> Fix mtd partitions udev rule
Resolves  #2172 rename '/dev/mtd/.tmp-c90:3' '/dev/mtd/' failed: Not a directory

closed 2017-09-08 by williamspatrick