openbmc/openbmc #2173: Failed to Boot Host - Host checkstop condition detected

gkeishin, 2017-08-19:
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
root@witherspoon:~#
```

`Host SOL log`
```
--== Welcome to Hostboot hostboot-be278ac/hbicore.bin ==--

  4.13093|secure|Booting in non-secure mode.
  4.13095|secure|SecureROM valid - enabling functionality
  5.70824|ISTEP  6. 5 - host_init_fsi
  5.83174|ISTEP  6. 6 - host_set_ipl_parms
  5.84033|ISTEP  6. 7 - host_discover_targets
  6.28603|HWAS|PRESENT> DIMM[03]=FFFF000000000000
  6.28605|HWAS|PRESENT> Proc[05]=C000000000000000
  6.28606|HWAS|PRESENT> TPM[31]=8000000000000000
  6.31128|ISTEP  6. 8 - host_update_master_tpm
  6.49605|SECURE|Security Access Bit> 0x0000000000000000
  6.49606|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
  6.49627|ISTEP  6. 9 - host_gard
  6.56652|ISTEP  6.10 - host_revert_sbe_mcs_setup
  6.57213|ISTEP  6.11 - host_start_occ_xstop_handler
  6.57755|ISTEP  6.12 - host_voltage_config
  6.61240|ISTEP  7. 2 - mss_volt
  6.88387|ISTEP  7. 3 - mss_freq
  7.01362|ISTEP  7. 4 - mss_eff_config
 10.34145|ISTEP  7. 5 - mss_attr_update
 10.35199|ISTEP  8. 1 - host_slave_sbe_config
 10.41094|ISTEP  8. 2 - host_setup_sbe
 10.41743|ISTEP  8. 3 - host_cbs_start
 10.43926|ISTEP  8. 4 - proc_check_slave_sbe_seeprom_complete
 20.05987|ISTEP  8. 5 - host_attnlisten_proc
 20.06110|ISTEP  8. 6 - host_p9_fbc_eff_config
 20.06624|ISTEP  8. 7 - host_p9_eff_config_links
 20.07746|ISTEP  8. 8 - proc_attr_update
 20.07940|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 20.12937|ISTEP  8.10 - proc_xbus_scominit
 23.34864|ISTEP  8.11 - proc_xbus_enable_ridi
 23.35360|ISTEP  9. 1 - fabric_erepair
 23.36591|ISTEP  9. 2 - fabric_io_dccal
 24.22425|ISTEP  9. 3 - fabric_pre_trainadv
 24.23421|ISTEP  9. 4 - fabric_io_run_training
 24.45038|ISTEP  9. 5 - fabric_post_trainadv
 24.45531|ISTEP  9. 6 - proc_smp_link_layer
 24.46057|ISTEP  9. 7 - proc_fab_iovalid
 24.48978|ISTEP  9. 8 - host_fbc_eff_config_aggregate
 24.49870|ISTEP 10. 1 - proc_build_smp
 24.71138|ISTEP 10. 2 - host_slave_sbe_update

```

`Jounrnald snippet`
```
Aug 03 16:01:25 witherspoon checkstop_app[2508]: Host checkstop condition detected
Aug 03 16:01:26 witherspoon phosphor-fru-fault-monitor[1168]: /xyz/openbmc_project/logging/entry/3
Aug 03 16:01:26 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Aug 03 16:01:26 witherspoon systemd[1]: Reached target Quiesce Target.
Aug 03 16:01:26 witherspoon systemd[1]: Reached target Host instance 0 crashed.
```

`Elog`
```
Entry : /xyz/openbmc_project/logging/entry/3
{
    "AdditionalData": [],
    "Id": 3,
    "Message": "org.open_power.Host.Error.Checkstop",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501776086287,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/2
{
    "AdditionalData": [],
    "Id": 2,
    "Message": "org.open_power.Host.Error.Checkstop",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501775988778,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [],
    "Id": 1,
    "Message": "org.open_power.Host.Error.Checkstop",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501775888224,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/4
{
    "AdditionalData": [],
    "Id": 4,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501776165866,
    "associations": []
}

```

`Driver Info`
```
root@witherspoon:~# cat /etc/timestamp
20170819012228
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Sat Aug 19 01:21:10 UTC 2017 armv6l GNU/Linux

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/686a9761": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-292-g92b1559"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
}
```

- [x] Journald 
[journald_v1.99.8-292-g92b1559_checkstop.txt](https://github.com/openbmc/openbmc/files/1236208/journald_v1.99.8-292-g92b1559_checkstop.txt)

geissonator, 2017-08-19:
There was a bad SBE image that got out to the lab, it would cause a checkstop when hostboot tried to update it.  This looks like that issue, need to manually update the SBE using cronus.

gkeishin, 2017-08-19:
On DD2 system this one worked good. on DD1 systems this looks broken. We would need a work-around or fix bad SBE image and verify.

geissonator, 2017-08-19:
I think Uma got you fixed up, in general host issues like this should be an issue on hostboot team repo.

closed 2017-08-19 by geissonator