openbmc/openbmc #2174: PNOR image upload Error footprint

gkeishin, 2017-08-19:
The PNOR update went good but keep seeing this footprint every upload of new image

`Journald snippet`
```
Aug 03 15:46:42 witherspoon phosphor-gevent[1328]: ::80e9:6976:8e91:f402%1995762460 - - [2017-08-03 15:46:42] "POST /login HTTP/1.1" 200 320 0.042475
Aug 03 15:46:43 witherspoon phosphor-version-software-manager[1118]: tar: short read
Aug 03 15:46:43 witherspoon phosphor-version-software-manager[1118]: Error No manifest file
Aug 03 15:46:43 witherspoon phosphor-version-software-manager[1118]: An error when reading the Manifest file.
Aug 03 15:46:43 witherspoon phosphor-fru-fault-monitor[1169]: /xyz/openbmc_project/logging/entry/6
Aug 03 15:46:43 witherspoon phosphor-version-software-manager[1118]: Error processing image
Aug 03 15:46:56 witherspoon phosphor-gevent[1328]: ::80e9:6976:8e91:f402%1995762460 - - [2017-08-03 15:46:56] "POST /upload/image HTTP/1.1" 200 169 13.556507
Aug 03 15:46:56 witherspoon phosphor-version-software-manager[1118]: Untaring
Aug 03 15:46:56 witherspoon systemd[1]: Created slice system-dropbear.slice.
Aug 03 15:46:56 witherspoon openpower-update-manager[1112]: Error in reading file
```
`Driver info`
```
/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/686a9761": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-292-g92b1559"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
}
```

- [x] Journald 
[journald_v1.99.8-292-g92b1559_upload_error.txt](https://github.com/openbmc/openbmc/files/1236317/journald_v1.99.8-292-g92b1559_upload_error.txt)

saqibkh, 2017-08-19:
Fixes one of the above mentioned error logs:

gtmills, 2017-08-21:
only happens on upload via REST. 
Works on downloadViaTFTP

gtmills, 2017-08-21:
The Image upload is closing the file twice. 
Triggering the inotify twice. 
https://github.com/openbmc/phosphor-rest-server/blob/18c3a2422fcdd619da7017c3f636623e0df470d0/module/obmc/wsgi/apps/rest_dbus.py#L673

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6146> Change ImageUpload to open file once
Resolves  #2174 PNOR image upload Error footprint

closed 2017-09-06 by williamspatrick