openbmc/openbmc #2175: Mbox service core-dump on BMC r/r during powering on

gkeishin, 2017-08-19:
1. Poweron Host 
2. r/r BMC when Host is Powering on
3.  BMC comes back and re IPL back Host to OS
4. OS booted

During booting back from BMC r/r , observed MBOX core dumped.

`Journald snippet`
```
Aug 19 15:08:03 witherspoon ipmid[1315]: Control Host work queue is empty!
Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Main process exited, code=dumped, status=11/SEGV
Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Unit entered failed state.
Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
````
```
root@witherspoon:~# systemctl status mboxd.service | cat
● mboxd.service - Phosphor MBOX Daemon
   Loaded: loaded (/lib/systemd/system/mboxd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-08-19 15:08:03 UTC; 1min 10s ago
 Main PID: 1535 (mboxd)
   CGroup: /system.slice/mboxd.service
           └─1535 /usr/sbin/mboxd --flash 64M --window-size 1M

Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Unit entered failed state.
Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
Aug 19 15:08:03 witherspoon systemd[1]: mboxd.service: Service hold-off time over, scheduling restart.
Aug 19 15:08:03 witherspoon systemd[1]: Stopped Phosphor MBOX Daemon.
Aug 19 15:08:03 witherspoon systemd[1]: Started Phosphor MBOX Daemon.
root@witherspoon:~#
```

`Driver Info`
```
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Sat Aug 19 01:21:10 UTC 2017 armv6l GNU/Linux

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/686a9761": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-292-g92b1559"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
}
```

- [x] Journald 
[journald_v1.99.8-292-g92b1559_mbox_coredump.txt](https://github.com/openbmc/openbmc/files/1236446/journald_v1.99.8-292-g92b1559_mbox_coredump.txt)

gkeishin, 2017-08-19:
Another same test results in this. but OS boot which is a good thing
However, we would expect application services to handle it better without core dumping or crashing out.

```
Aug 19 15:24:41 witherspoon ipmid[1313]: Host control timeout hit!
Aug 19 15:24:41 witherspoon ipmid[1313]: Timer expired
Aug 19 15:24:41 witherspoon phosphor-host-check[1360]: Host control signal values
Aug 19 15:24:41 witherspoon phosphor-host-check[1360]: Host is not running!
Aug 19 15:24:41 witherspoon systemd[1]: Started Check Host0 status on BMC reset.
Aug 19 15:24:41 witherspoon systemd[1]: Reached target Host0 running after reset.
Aug 19 15:24:41 witherspoon systemd[1]: Starting Phosphor Host State Manager...
Aug 19 15:24:41 witherspoon systemd[1]: Reached target Host0 (Reset Check).
Aug 19 15:24:42 witherspoon phosphor-host-state-manager[1371]: Initial Host State will be Off
Aug 19 15:24:42 witherspoon phosphor-host-state-manager[1371]: Host State transaction request
Aug 19 15:24:42 witherspoon systemctl[1341]: Job for obmc-chassis-poweron@0.target canceled.
Aug 19 15:24:42 witherspoon phosphor-host-state-manager[1371]: External request to reset reboot count
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 19 15:24:43 witherspoon systemd[1]: Stopped Start chassis0 on after BMC reset.
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Unit entered failed state.
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Failed with result 'exit-code'.
Aug 19 15:24:43 witherspoon systemd[1]: Started Phosphor Host State Manager.
Aug 19 15:24:43 witherspoon systemd[1]: Created slice system-op\x2docc\x2ddisable.slice.
Aug 19 15:24:43 witherspoon systemd[1]: Stopped target Chassis0 (Reset Check).
Aug 19 15:24:43 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Aug 19 15:24:43 witherspoon occ-active.sh[1374]: /usr/bin/occ-active.sh: line 16:  1375 Done                    busctl tree org.open_power.OCC.Control --list
Aug 19 15:24:43 witherspoon occ-active.sh[1374]:       1376 Done(1)                 | grep occ
Aug 19 15:24:43 witherspoon occ-active.sh[1374]:       1377 Segmentation fault      | xargs -n1 -I{} busctl set-property org.open_power.OCC.Control {} org.open_power.OCC.Status OccActive b $value
Aug 19 15:24:43 witherspoon systemd[1]: Created slice system-obmc\x2dled\x2dgroup\x2dstop.slice
```

```
root@witherspoon:~# systemctl status op-reset-chassis-on@0.service | cat
● op-reset-chassis-on@0.service - Start chassis0 on after BMC reset
   Loaded: loaded (/lib/systemd/system/op-reset-chassis-on@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2017-08-19 15:24:43 UTC; 2min 13s ago
  Process: 1341 ExecStart=/bin/systemctl start obmc-chassis-poweron@%i.target (code=exited, status=1/FAILURE)
 Main PID: 1341 (code=exited, status=1/FAILURE)

Aug 19 15:24:34 witherspoon systemd[1]: Started Start chassis0 on after BMC reset.
Aug 19 15:24:42 witherspoon systemctl[1341]: Job for obmc-chassis-poweron@0.target canceled.
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 19 15:24:43 witherspoon systemd[1]: Stopped Start chassis0 on after BMC reset.
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Unit entered failed state.
Aug 19 15:24:43 witherspoon systemd[1]: op-reset-chassis-on@0.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

geissonator, 2017-08-20:
George, can you be more specific on where the host is in the boot when you inject the bmc reboot?  Our support for a bmc reboot when the host is booting is lacking a bit.  But we definitely want to make sure if you're up to petitboot, that a reboot of the BMC does not impact the OS at all.

gkeishin, 2017-08-20:
Yeah this one was done manually and I didn't wait till petitboot to do BMC r/r. This more like it just crossed istep 21 and started loading petitboot. I ll try out some more on it and see if I can provide more data to debug.

geissonator, 2017-08-20:
Heh, I definitely know we don't really handle that scenario very well so don't spend too much time on it, need to dig into the code a bit and probably fix some things up with this bug.  This is def an error path, so will mark this 5.1.

gkeishin, 2017-08-20:
Sure thing then..

gkeishin, 2017-09-14:
I haven't seen this in the later build and as well looks clean in Master. Closing it up..

closed 2017-09-14 by gkeishin