openbmc/openbmc #2177: Disabling auto reboots with host-failure-reboots@0.service doesn't work first time

geissonator, 2017-08-19:
We tell people to run "systemctl stop host-failure-reboots@0.service" to disable automatic reboots of the host.  But since this is a service, if it's already stopped, it wont do anything when you run it.  You need to first start it, then stop it.  I tried a --force but didn't help.  We may need 2 services here, one to disable, one to enable, where you start both of them.

```
[witherspoon]# systemctl stop --force host-failure-reboots\@0.service [witherspoon]# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b true
[witherspoon]# systemctl stop host-failure-reboots\@0.service [witherspoon]# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b true
[witherspoon]# systemctl start host-failure-reboots\@0.service [witherspoon]# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b true
[witherspoon]# systemctl stop host-failure-reboots\@0.service [witherspoon]# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b false
```

geissonator, 2017-08-19:
Workaround is to just run the busctl command directly:

busctl set-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot b false

Or just start and then stop the service.

spinler, 2017-11-15:
I meant to have this service started on the way to standby, but I forgot to link it in to obmc-standby.target.wants

gtmills, 2018-07-18:
@geissonator Can this be closed since we have a workaround? Do we need to document it anywhere?

spinler, 2018-07-18:
I think the problem is I had forgotten to start this service on the way to standby, so trying to stop an already stopped service didn't do anything.

gtmills, 2018-07-18:
@spinler Can we just close then?

closed 2019-10-28 by stale[bot]