openbmc/openbmc #2179: DCMI request failed because: Unspecified error (ff)

gkeishin, 2017-08-19:
Commnad 
```
ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx dcmi set_asset_tag bsykU
```

`Journald snippet`
```
Aug 19 19:57:08 witherspoon netipmid[1318]: IPMI GROUP EXTENTIONS
Aug 19 19:57:08 witherspoon netipmid[1318]: >> closeSession
Aug 19 19:57:08 witherspoon netipmid[1318]: << closeSession
Aug 19 19:57:08 witherspoon netipmid[1318]: >> GetChannelCapabilities
Aug 19 19:57:08 witherspoon netipmid[1318]: << GetChannelCapabilities
Aug 19 19:57:08 witherspoon netipmid[1318]: >> openSession
Aug 19 19:57:08 witherspoon netipmid[1318]: << openSession
Aug 19 19:57:08 witherspoon netipmid[1318]: >> RAKP12
Aug 19 19:57:08 witherspoon netipmid[1318]: << RAKP12
Aug 19 19:57:08 witherspoon netipmid[1318]: >> RAKP34
Aug 19 19:57:08 witherspoon netipmid[1318]: << RAKP34
Aug 19 19:57:08 witherspoon netipmid[1318]: >> setSessionPrivilegeLevel
Aug 19 19:57:08 witherspoon netipmid[1318]: << setSessionPrivilegeLevel
Aug 19 19:57:08 witherspoon netipmid[1318]: E> Table:: Command Not found: 0xb03e
Aug 19 19:57:09 witherspoon netipmid[1318]: AssetTag property is not populated
Aug 19 19:57:09 witherspoon netipmid[1318]: The operation failed internally.
Aug 19 19:57:09 witherspoon netipmid[1318]: AssetTag property is not populated
Aug 19 19:57:09 witherspoon netipmid[1318]: The operation failed internally.
Aug 19 19:57:10 witherspoon netipmid[1318]: IPMI GROUP EXTENTIONS
Aug 19 19:57:10 witherspoon netipmid[1318]: >> closeSession
Aug 19 19:57:10 witherspoon netipmid[1318]: << closeSession
Aug 19 19:57:10 witherspoon netipmid[1318]: >> GetChannelCapabilities
Aug 19 19:57:10 witherspoon netipmid[1318]: << GetChannelCapabilities
Aug 19 19:57:10 witherspoon netipmid[1318]: >> openSession
Aug 19 19:57:10 witherspoon netipmid[1318]: << openSession
Aug 19 19:57:10 witherspoon netipmid[1318]: >> RAKP12
Aug 19 19:57:10 witherspoon netipmid[1318]: << RAKP12
Aug 19 19:57:10 witherspoon netipmid[1318]: >> RAKP34
Aug 19 19:57:10 witherspoon netipmid[1318]: << RAKP34
Aug 19 19:57:10 witherspoon netipmid[1318]: >> setSessionPrivilegeLevel
Aug 19 19:57:10 witherspoon netipmid[1318]: << setSessionPrivilegeLevel
Aug 19 19:57:10 witherspoon netipmid[1318]: E> Table:: Command Not found: 0xb03e
Aug 19 19:57:10 witherspoon netipmid[1318]: AssetTag property is not populated
Aug 19 19:57:10 witherspoon netipmid[1318]: The operation failed internally.
```

gkeishin, 2017-08-19:
@tomjoseph83 anything on this

sivassrr, 2017-08-20:
Could you add entire journalctl as well.

gkeishin, 2017-08-20:
We don't need entire journald to debug this. It is pervasive and the error needed to debug is in the above snippet.

sivassrr, 2017-08-20:
May want to stick with the same format, Changes that we may get other leads based on journalctl.

tomjoseph83, 2017-08-23:
@gkeishin Do you have the build this issue can be reproduced on? It looks like a case of inventory manager was down.

gkeishin, 2017-08-24:
Well it's seen in few previous master build .. but just to mentioned today to recreate on the newer build as well

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-39"
VERSION_ID="v1.99.9-39-g576714d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-39"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Thu Aug 24 03:21:19 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```

tomjoseph83, 2017-09-18:
```
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r3-0-gdfda752"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r3"
```
This issue is not reproducible on the latest GHE release. We can close it.

gkeishin, 2017-09-18:
Sure good thing..

closed 2017-09-18 by gkeishin

rfrandse, 2018-02-06:
<https://gerrit.openbmc-project.xyz/8852> busybox: Enable expand(1) utility
Resolves:  #2179 DCMI request failed because: Unspecified error (ff)