openbmc/openbmc #2180: BMC r/r when in petitboot

gkeishin, 2017-08-20:
- Boot Host
- r/r BMC when on Petitboot

Post RR. BMC comes up and host send down poweroff 
```
.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .[  192.821279] reboot: Power down
[  329.023312280,5] OPAL: Shutdown request type 0x0...
Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .

```

```
 Petitboot (v1.5.1-p641e963)                  PNW51            SN-W51
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
  [Disk: sda2 / 63fb36f8-b3e6-4f50-ae8e-50fb8169e667]
    Ubuntu, with Linux 4.10.0-28-generic (recovery mode)
    Ubuntu, with Linux 4.10.0-28-generic
    Ubuntu

  System information
  System configuration
  System status log
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell

 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
 Enter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
 Booting in 1 sec: [sda2]
```

`Journald snippet`
```
Aug 20 18:24:32 witherspoon phosphor-host-state-manager[1368]: Host State transaction request
Aug 20 18:24:33 witherspoon systemctl[1340]: Job for obmc-chassis-poweron@0.target canceled.
Aug 20 18:24:33 witherspoon systemctl[1367]: Job for obmc-host-start@0.target canceled.
Aug 20 18:24:33 witherspoon phosphor-host-state-manager[1368]: External request to reset reboot count
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 20 18:24:33 witherspoon systemd[1]: Stopped Start chassis0 on after BMC reset.
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Unit entered failed state.
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Failed with result 'exit-code'.
Aug 20 18:24:33 witherspoon systemd[1]: phosphor-reset-host-running@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 20 18:24:33 witherspoon systemd[1]: phosphor-reset-host-running@0.service: Unit entered failed state.
Aug 20 18:24:33 witherspoon systemd[1]: phosphor-reset-host-running@0.service: Failed with result 'exit-code'.
```

```
Aug 20 18:25:13 witherspoon systemd[1]: Starting Reload mboxd during power on...
Aug 20 18:25:14 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Main process exited, code=killed, status=15/TERM
Aug 20 18:25:14 witherspoon systemd[1]: Stopped Soft power off of the host.
Aug 20 18:25:14 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Unit entered failed state.
Aug 20 18:25:14 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Failed with result 'signal'.
```

- [x] Journald 
[rr-bmc-petitboot-v1.99.8-294-g31faf40.txt](https://github.com/openbmc/openbmc/files/1237263/rr-bmc-petitboot-v1.99.8-294-g31faf40.txt)

gkeishin, 2017-08-20:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-294"
VERSION_ID="v1.99.8-294-g31faf40"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-294"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Sun Aug 20 12:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```

geissonator, 2017-08-20:
Initial detection went good:
```
Aug 20 18:24:26 witherspoon ipmid[1320]: SMS Attention asserted
Aug 20 18:24:26 witherspoon phosphor-host-check[1343]: Host control signal values
Aug 20 18:24:26 witherspoon phosphor-host-check[1343]: Host is running!
Aug 20 18:24:26 witherspoon systemd[1]: Started Check Host0 status on BMC reset.
Aug 20 18:24:26 witherspoon systemd[1]: Reached target Host0 running after reset.
Aug 20 18:24:26 witherspoon systemd[1]: Started Start host0 running after BMC reset.
Aug 20 18:24:26 witherspoon systemd[1]: Starting Phosphor Host State Manager...
```

but then something caused a power off request, which starts to hose everything
```
Aug 20 18:24:31 witherspoon systemd[1]: Started Wait for /org/openbmc/control/chassis0.
Aug 20 18:24:31 witherspoon phosphor-host-state-manager[1368]: Initial Host State will be Off
Aug 20 18:24:31 witherspoon phosphor-ledmanager[1075]: Asserting LEDs
Aug 20 18:24:32 witherspoon systemd[1]: Started Assert power_on LED.
Aug 20 18:24:32 witherspoon phosphor-host-state-manager[1368]: Host State transaction request
Aug 20 18:24:33 witherspoon systemctl[1340]: Job for obmc-chassis-poweron@0.target canceled.
Aug 20 18:24:33 witherspoon systemctl[1367]: Job for obmc-host-start@0.target canceled.
Aug 20 18:24:33 witherspoon phosphor-host-state-manager[1368]: External request to reset reboot count
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 20 18:24:33 witherspoon systemd[1]: Stopped Start chassis0 on after BMC reset.
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Unit entered failed state.
Aug 20 18:24:33 witherspoon systemd[1]: op-reset-chassis-on@0.service: Failed with result 'exit-code'.
Aug 20 18:24:33 witherspoon systemd[1]: phosphor-reset-host-running@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 20 18:24:33 witherspoon systemd[1]: phosphor-reset-host-running@0.service: Unit entered failed state.
```

mdmillerii, 2017-08-20:
The first messages are for `Host0` with capitol `H`.
```
Aug 20 18:24:26 witherspoon systemd[1]: Started Check Host0 status on BMC reset.
Aug 20 18:24:26 witherspoon systemd[1]: Reached target Host0 running after reset.
```
The last is for `host0` with lower case `h`.
```
Aug 20 18:24:26 witherspoon systemd[1]: Started Start host0 running after BMC reset.
```

and then
```
Aug 20 18:24:31 witherspoon phosphor-host-state-manager[1368]: Initial Host State will be Off
```
Is this relevant? 

PS I don't see an instance name in the state manager, just the PID in the log.  Might be hard to debug later if we have multiple instances.

geissonator, 2017-08-21:
Milton, the messages above just come from the Description field in the service files.  Be nice if it was consistent but doesn't really matter.

I believe I have a potential fix for this up here: https://gerrit.openbmc-project.xyz/#/c/6143/

The issue is that the phosphor-discover-system-state application has a race condition with the reset/reload targets.  Sometimes it correctly runs after the state has been discovered on the r/r, and sometimes it runs before.  This patch set will disable it from running when we're in a r/r scenario (i.e. the chassis power and/or host is running).  In those cases we want the BMC to just adjust to the state of the system and leave it as-is.

gkeishin, 2017-08-21:
We observe this in newer build too.. so the new fix would be interesting to verify..

rfrandse, 2017-08-21:
<https://gerrit.openbmc-project.xyz/6143> Don't apply restore policy if system is on
Resolves  #2180 BMC r/r when in petitboot - Shutdown Host post r/r

geissonator, 2017-08-23:
# **STATUS**
- 08/23: Code up and tested, ready for peer reviews

closed 2017-08-24 by williamspatrick