openbmc/openbmc #2181: OS Bootprogress state is Unspecified when OS is booted

gkeishin, 2017-08-20:
`Before`
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
        "AttemptsLeft": 3,
        "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
        "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }
}

```
`OS is booted`

```
Ubuntu 17.04 w51L hvc0

w51L login:
```

`Issue BMC r/r`

```
BMC rest and comes back up but the Bootprogress is off`
```

`Post RR`
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
        "AttemptsLeft": 3,
        "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified",
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
        "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }
}
```
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-294"
VERSION_ID="v1.99.8-294-g31faf40"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-294"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Sun Aug 20 12:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```

- [x] Journald 
[rr-bmc-osbooted-v1.99.8-294-g31faf40.txt](https://github.com/openbmc/openbmc/files/1237286/rr-bmc-osbooted-v1.99.8-294-g31faf40.txt)

dhruvibm, 2017-08-22:
@gkeishin was the bootprogress state persisted across r/r in older drivers?

gkeishin, 2017-08-22:
I don't think we saw boot progress persisted in older ORG...

dkodihal, 2017-08-22:
@gkeishin is this a new requirement then? Can we capture what is a bug and what is a new requirement?

If we haven't regressed in any way, I don't think this is a phase 5 @geissonator. Although the title of the bug doesn't sound like it, when you poweron a box to the OS, the boot progress state is updated correctly; @gkeishin has also confirmed this to me. If this is purely a R/R case, I don't think this is phase 5, or a bug even if that was never implemented with the old sensors.

gkeishin, 2017-08-22:
In old ORG the boot progress will correct show if the OS was Off or on.. so we didn't really have a test to check the persistence as such as it was kind of tested in all state management context for all then poweron/off/ reset we got in automation.. so we need this one ,

dkodihal, 2017-08-22:
I'm not saying you don't need it - is this a phase 5 bug, or a new requirement?

geissonator, 2017-08-22:
Clarified in issue title that this is after a BMC reboot with the host up.  That's a 5.1 type thing so moved phase.

dhruvibm, 2017-08-24:
- [ ] Support  persistence of boot progress value

rfrandse, 2017-09-19:
<https://gerrit.openbmc-project.xyz/6741> Add persistancy for boot progress and os status
Resolves  #2181 Bootprogress state is Unspecified after bmc reboot when OS is booted

dhruvibm, 2017-09-20:
# **STATUS**
9/20: Ready for merge review

closed 2017-09-27 by williamspatrick