openbmc/openbmc #2182: Romulus: VUART mirroring is broken

mine260309, 2017-08-21:
The VUART is mirrored to physical UART on Romulus.
The mirror was on ttyS0, but now the ttyS0 is actually ttyVUART, probably due to kernel's update.
The physical UART is now on ttyS1 instead, so the config file (obmc-console.conf) shall be updated to use ttyS1:
```
lpc-address = 0x3f8
sirq = 4
local-tty = ttyS0  # This needs change to ttyS1
local-tty-baud = 115200
```

I have confirmed thsi on Romulus, and I think other P9 systems have this issue as well.

mdmillerii, 2017-08-21:
@shenki @amboar This a race allowing `vuart` to steal `ttyS0` because it is not listed in the `serial` `/alias`.

The [vuart driver](https://github.com/openbmc/linux/blob/dev-4.10/drivers/tty/serial/8250/8250_aspeed_vuart.c#L253-L255) will look up a serial alias to get the line number resulting in the `ttyS0` number.

Seems like openbmc/linux@10b274619befc2469ddf7eea5ebd6595079b2ff6 needs to add
```
     serial5 = &vuart;
```
to `/aliases`.

shenki, 2017-08-23:
That's an acceptable workaround. @mine260309 has volunteered to send a patch

mine260309, 2017-08-23:
@mdmillerii @amboar The change `serial5 = &vuart;` does not fix the issue, instead, it cause the device `/dev/ttyVUART` disappear, which breaks the VUART function.
I will investigate further. But if you know what's wrong please let me :)

amboar, 2017-08-24:
@mine260309 unfortunately I'm just casually observing at the moment, I don't have any good insights.

williamspatrick, 2017-08-24:
I wonder if the same problem exists on Witherspoon then.  I think I saw ttyS0 being symlinked to ttyVUART today.

mine260309, 2017-08-24:
@mdmillerii @amboar The change `serial5 = &vuart;` actually works.
I was meeting the issue because the `.config` somehow disables `CONFIG_SERIAL_8250_ASPEED_VUART` and thus the `vuart` device is not probed at all.

@williamspatrick Yeah, I believe Witherspoon should have the same issue. `ttyVUART` is symlinked to `ttyS0`, exactly the same as Romulus.

I will send a patch to fix the issue.

mine260309, 2017-08-25:
Patch: http://patchwork.ozlabs.org/patch/805357/

closed 2017-08-31 by mine260309