openbmc/openbmc #2185: [xCAT] Setting VLAN inband is OK, turning VLAN off inband does not work

whowutwut, 2017-08-21:
Testing use the GHE build from 08/13 that has VLAN tagging support in.  FYI @ratagupt 

## Setting does work

Did...
```
[root@frame6u05 ~]#ipmitool lan set 1 vlan id 123
[root@frame6u05 ~]# ipmitool  lan set 1 access on
IPMI command failed: Invalid command
Unable to Get Channel Access(non-volatile) for channel 1
[root@frame6u05 ~]# ipmitool raw 0x06 0x40
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
```

Even though it has some ugly responses, the network configuration does in fact change and things look good.  
```
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet6 addr: fe80::72e2:84ff:fe14:280d%1260204944/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4154358 errors:0 dropped:0 overruns:0 frame:0
          TX packets:304443 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:449565210 (428.7 MiB)  TX bytes:19449748 (18.5 MiB)
          Interrupt:20

eth0.123  Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:10.6.5.253  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::72e2:84ff:fe14:280d%1260204944/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:5852 (5.7 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1260204944/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1655052 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1655052 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:125784624 (119.9 MiB)  TX bytes:125784624 (119.9 MiB)

root@witherspoon:/etc/systemd/network# ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 70:e2:84:14:28:0d brd ff:ff:ff:ff:ff:ff
    inet6 fe80::72e2:84ff:fe14:280d/64 scope link
       valid_lft forever preferred_lft forever
4: eth0.123@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue qlen 1000
    link/ether 70:e2:84:14:28:0d brd ff:ff:ff:ff:ff:ff
    inet 10.6.5.253/8 brd 10.255.255.255 scope global eth0.123
       valid_lft forever preferred_lft forever
    inet6 fe80::72e2:84ff:fe14:280d/64 scope link
       valid_lft forever preferred_lft forever
```

## Unset VLAN does not work

Did the following: 
```
[root@frame6u05 ~]# ipmitool  lan set 1 vlan id off
[root@frame6u05 ~]# ipmitool raw 0x06 0x40
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
```

The raw command reboots the eth0 interface, but none of the IP configuration files are cleared out and the VLAN tag is still active:
```
[root@frame6u05 ~]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address              : 10.6.5.253
Subnet Mask             : 255.0.0.0
MAC Address             : 70:e2:84:14:28:0d
Default Gateway IP      : 96.228.0.0
802.1q VLAN ID          : 123
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```

whowutwut, 2017-08-21:
I need to move this node to a Tagged environment to validate if I can ping when it's turned on, but I have to reset the BMC first...

whowutwut, 2017-08-21:
@ratagupt  I'm unable to set IP address now after trying to turn off VLAN on the BMC...  

`vlan id off` does not work as stated above.. .so I cleared out the `/etc/systemd/network/*` directory and added the following:
```
[Match]
Name=eth0
[Network]
DHCP=true
[DHCP]
ClientIdentifier=mac
```
then rebooted the BMC...   

It comes back up looking good... but now I can't set static IP address using any method...`busctl`, inband `ipmitool` calls... Always get the following message in the journalctl logs...

```
Aug 21 17:47:39 witherspoon phosphor-network-manager[1172]: DHCP enabled on the interface
```

mdmillerii, 2017-08-21:
@whowutwut you didn't say what name you used for your configuration file.  `systemd-netoworkd` assembles all configuration files from `/usr/lib`, `/run`, and `/etc` , and sorts them by name.  It then iterates through them, ignoring any like named files in a lower priority directory.

Once a network is configured, either statically before its starts or dynamically from any rule, all additional rules will be ignored for that interface.

Since the lib directory has a file named `default` that configures `DHCP`, your override needs to have a name that sorts before that like `bmc-eth0`; it can't be `eth0` or `vlan0` as those are after default.

Not sure if this is your issue.

whowutwut, 2017-08-21:
@mdmillerii Thanks for you reply!  We have been trying NOT to manually modify any system files in the BMC...   and instead using the `busctl` command to manually set the configuration for those systems that do not have a DHCP server, or using the REST API to set the configuration.  We prefer to use the REST API all the time. 

In testing VLAN, I used `ipmitool` commands inband from the host OS, since this is a requirement that @ratagupt  knows about.  This causes all the configuration in `/etc/systemd/network` to look correct as well as `ifconfig` and `ip addr show` looked OK.  I could not ping the node because the node currently is not behind a switch that has VLAN tagging enabled.  That is my next step. 

However, trying to undo the VLAN Settings and going back to DHCP, I had to manually run on the BMC because the IP address could not be used with the tagging enabled.  Once I set it back to DHCP, it seems like the DHCPEnabled=1 and I could not run and commands to override this until I set it to DHCPEnabled=0, then it would work.  

Maybe this is by design, but would cause some issues if we do lose network connectivity to the BMC, and we serial cable into the node .. or ssh IPV6... would be helpful to know the method for doing a "factory Reset" via commands line in the BMC 

@zet809 FYI

ratagupt, 2017-08-22:
@whowutwut :- It was by design that if interface is configured as Dynamic,User can not set the IP address on it.
What I feel that If interface is configured as DHCP and doesn't get the IP from DHCP Server then It should switch to Zero Config IP.
User can access the system through zero config IP and set the desired IP address on it.
Seems to me that system is not coming up with zero config ip.

Now your second issue is after VLAN id set off through IPMI doesn't work.It is related to 
https://github.com/openbmc/openbmc/issues/2150
Fix is under review.

whowutwut, 2017-08-22:
@ratagupt so for the first issue, is that a change you will make? To switch to zero config IP?

ratagupt, 2017-08-22:
Yes,You will get it through ..
https://gerrit.openbmc-project.xyz/#/c/6079/2

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6078> vlan:Remove the multiple implementation of sdbusplus object
Resolves  #2185 [xCAT] Setting VLAN inband is OK, turning VLAN off inband does not work

whowutwut, 2017-08-23:
@ratagupt @bradbishop  

I think we need some documentation or examples about how users should be setting the IP address and manipulating IP address...  or translating ipmitool commands to the REST commands...  it's not intuitive (maybe it's bugs) but without some documentation, developer teams are free to change the default behavior at any time, and or the externals at any time...  Without an external, consumers can't code against the API independent on the development schedule..... 

For example, using ipmitool, we can do the following: 
```
ipmitool lan set <ch> ipsrc static 
ipmitool lan set <ch> ipaddr <IP> 
ipmitool lan set <ch> netmask <NETMASK>
ipmitool lan set <ch> defgw ipaddr <GW IP> 
ipmitool lan set <ch> vlan id <VLAN ID> 
ipmitool lan set <ch> access on
```

Any of these can be called and it will change only the attributes called... .  with REST, it's completely different.   From what I have gathered and tested.... 

1. `ipmitool lan set <ch> ipsrc static ` maps to:  " send data=0 to `<interface>/attr/DCHPEnabled` endpoint "
2. You CANNOT modify IP/Netmask/Gateway on existing network objects, (BUG) rest return 200 OK ...  we need to create a NEW object for each change and now the caller must retrieve the object and overlay the changes... for example, just changing the IP required a GET, ensuring that we obtain the correct inet config.... create a new object and overlay the change on top.... then after it auto applies, delete the old object. 
3. No need to run `access on` as it now auto applies when REST, but not with in-band IPMI...
4. VLAN is set by creating a NEW object against the `<interface>_<VLANID>` (i.e eth0_123) for VLAN 123...

closed 2017-08-23 by williamspatrick