openbmc/openbmc #2186: [xCAT] Change Power Restore Policy to default: AlwaysOff

whowutwut, 2017-08-21:
Change power policy to Always Off by default.  

In a recent firmware, after applying power to the BMC the host automatically booted.  I'm not sure where the use case was to set the default Power Policy to "Restore" but this default will change behavior from other IBM OpenPower systems (P8) and what will happen with Witherspoon machines. 

I know this project is open source in general, so if there is a reason why Power Policy is Restore, let's discuss.  We will also look in to controlling this from xCAT so that we can change this behavior, but I'm wondering if other management utilities writing against OpenBMC also do not expect hosts to power back on after a power outtage.  

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Power/RestorePolicy.interface.yaml

geissonator, 2017-08-21:
Commercial machines usually default to "off unless told otherwise" so we'll go with this as the default.

rfrandse, 2017-08-21:
<https://gerrit.openbmc-project.xyz/6142> Default PowerRestorePolicy to always off
Resolves  #2186 [xCAT] Change Power Restore Policy to default: AlwaysOff

closed 2017-08-23 by williamspatrick