openbmc/openbmc #2188: phosphor-fan-control[2065]: Unable to find property

gkeishin, 2017-08-21:
On the latest Master on boot see the following errors. The Host did boot Good on DD2 system.

```
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Unable to find property:
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Error in mapper call to get service name
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: The operation failed internally.
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Unable to find property:
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Error in mapper call to get service name
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: The operation failed internally.
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Unable to find property:
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Error in mapper call to get service name
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: The operation failed internally.
Aug 21 18:57:19 witherspoon phosphor-fan-control[2065]: Unable to find property:
Aug 21 18:57:20 witherspoon phosphor-fan-control[2065]: Error in mapper call to get service name
Aug 21 18:57:20 witherspoon phosphor-fan-control[2065]: The operation failed internally.
Aug 21 18:57:20 witherspoon phosphor-fan-control[2065]: Unable to find property:

```

- [x] Journald 
[mapper_fan_error_v1.99.8-303-gc5359f4.txt](https://github.com/openbmc/openbmc/files/1239849/mapper_fan_error_v1.99.8-303-gc5359f4.txt)

gkeishin, 2017-08-21:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-303"
VERSION_ID="v1.99.8-303-gc5359f4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-303"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Mon Aug 21 01:27:02 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

gkeishin, 2017-08-21:
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan0": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan0"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan1": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan1"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan2": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan2"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan3": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": "fan3"
    },
```
I see the fan are up and functional.. so there is probably some other issue here.

msbarth, 2017-08-21:
phosphor-fan-control doesnt do anything with fan functional/present states
These journal entries are for properties given within the event yaml that were not found to be used in any set speed event. In this case, its likely due to "missing" cores and/or dimms since there is a single event yaml for all system configurations, all items within a set speed event group need to be listed but may not be populated.

The journal entry exists since these are user provided into the yaml in case of typos, etc.. were entered for any specific speed event

msbarth, 2017-08-22:
Updated to create informational log entries for properties not found.

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6170> Create informational logs for properties not found
Resolves  #2188 phosphor-fan-control[2065]: Unable to find property

closed 2017-08-25 by williamspatrick