openbmc/openbmc #2189: occ-active.sh script segmentation fault

gkeishin, 2017-08-21:
On latest Master 984 build:
- [x] At BMC standby issue r/r
and we we seen this in log

```
Aug 21 19:21:41 witherspoon systemd[1]: Created slice system-op\x2docc\x2ddisable.slice.
Aug 21 19:21:41 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Aug 21 19:21:41 witherspoon systemd[1]: Starting Wait for /xyz/openbmc_project/led/groups/power_on...
Aug 21 19:21:41 witherspoon phosphor-host-state-manager[1237]: External request to reset reboot count
Aug 21 19:21:42 witherspoon occ-active.sh[1243]: /usr/bin/occ-active.sh: line 16:  1245 Done                    busctl tree org.open_power.OCC.Control --list
Aug 21 19:21:42 witherspoon occ-active.sh[1243]:       1246 Done(1)                 | grep occ
Aug 21 19:21:42 witherspoon occ-active.sh[1243]:       1247 Segmentation fault      | xargs -n1 -I{} busctl set-property org.open_power.OCC.Control {} org.open_power.OCC.Status OccActive b $value
Aug 21 19:21:42 witherspoon systemd[1]: Created slice system-op\x2dpower\x2dstop.slice.
```

```
busctl tree org.open_power.OCC.Control --list | grep occ | xargs -n1 -I{} \
    busctl set-property org.open_power.OCC.Control {} \
org.open_power.OCC.Status OccActive b $value
```
https://github.com/openbmc/openbmc/blob/82cad8431dcba9c19daeb2cc2b2f94ac580311e9/meta-openbmc-machines/meta-openpower/common/recipes-phosphor/occ/openpower-occ-control/occ-active.sh

This failed.

```
root@witherspoon:~# busctl tree org.open_power.OCC.Control --list | cat
/
/org
/org/open_power
/org/open_power/control
root@witherspoon:~#
```

need to take into consideration mechanism to not fail if the process doesn’t exist, or if the property is not available.

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-303"
VERSION_ID="v1.99.8-303-gc5359f4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-303"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1 Mon Aug 21 01:27:02 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```

- [x] Journald 
[Segmentation_occ_v1.99.8-303-gc5359f4.txt](https://github.com/openbmc/openbmc/files/1239921/Segmentation_occ_v1.99.8-303-gc5359f4.txt)

geissonator, 2017-08-21:
There are scenarios where the occ-active.sh script may get called but the openpower-occ-control service is not available (or the occ status objects are not created yet).  In those cases we just want this script to be a no-op.

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6144> Exit gracefully if occ status object not available
Resolves  #2189 occ-active.sh script segmentation fault

geissonator, 2017-08-23:
# **STATUS**
- 08/23: Code up, tested, through a few reviews, ready for re-review by peers

closed 2017-08-29 by williamspatrick