openbmc/openbmc #2191: Unable to getcfam second processor after chassis power on

geissonator, 2017-08-21:
We put some fixes in https://gerrit.openbmc-project.xyz/#/c/5862/ which started doing the second proc chip hub scan in the chassis poweron sequence.  When we did this though, we found we could not do getcfam's on the second proc chip.  Need to get to the bottom of this because certain inits are run against it during the host_start procedure.

```
[witherspoon]# obmcutil chassison
[witherspoon]# ./pdbg getcfam -p0 0x2918
p0:0x2918 = 0xeeeef304
[witherspoon]# ./pdbg getcfam -p1 0x2918
No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
Alternatively run ./pdbg -a probe to get a list of all valid targets
```

cbostic, 2017-08-22:
++++++++++++++++++++++++++++++++++++++

#STATUS

8/21: Able to recreate on w51 with my test kernel. 
Appears that running back to back base plus hub scans from command line gets out of this fail mode.  i.e.: 
```echo 1 > /sys/devices/platform/gpio-fsi/fsi0/rescan; echo 1 > /sys/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:0a/fsi1/rescan
```
An interim solution may be to run these two scans back to back instead of just the hub scan. 

Still need to investigate why a full scan is required...
++++++++++++++++++++++++++++++++++++++

closed 2017-08-24 by williamspatrick

geissonator, 2017-08-25:
After going through the different services that we run to power on the chassis, we found that this change was causing the FSI clock fail - https://gerrit.openbmc-project.xyz/#/c/5628/

This change came from our power engineer, for now we've commented out the change causing the problem and will follow up in #2204 .