openbmc/openbmc #2192: Boot Device Option set incorrectly ?

gkeishin, 2017-08-22:
Expected to have `Force Boot into BIOS Setup` as per the test we got. 

```
ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx raw 0x0 0x8 0x05 0x80 0x18 0x00 0x00 0x00 2
``
```
$ ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx chassis bootparam get 5 2
Boot parameter version: 1
Boot parameter 5 is valid/unlocked
Boot parameter data: 8014000000
 Boot Flags :
   - Boot Flag Valid
   - Options apply to only next boot
   - BIOS PC Compatible (legacy) boot
   - Boot Device Selector : Force Boot from CD/DVD
   - Console Redirection control : System Default
   - BIOS verbosity : Console redirection occurs per BIOS configuration setting (default)
   - BIOS Mux Control Override : BIOS uses recommended setting of the mux at the end of POST
``

```
Aug 03 17:16:45 witherspoon netipmid[2864]: IPMI GROUP EXTENTIONS
Aug 03 17:16:45 witherspoon netipmid[2864]: IPMI GET_SYS_BOOT_OPTIONS
Aug 03 17:16:45 witherspoon netipmid[2864]: IPMID boot option property get: {ONETIME}.
Aug 03 17:16:45 witherspoon netipmid[2864]: BootPolicy is [ONETIME]
Aug 03 17:16:45 witherspoon netipmid[2864]: >> closeSession
Aug 03 17:16:45 witherspoon netipmid[2864]: << closeSession
Aug 03 17:16:45 witherspoon netipmid[2864]: >> GetChannelCapabilities
Aug 03 17:16:45 witherspoon netipmid[2864]: << GetChannelCapabilities
Aug 03 17:16:45 witherspoon netipmid[2864]: >> openSession
Aug 03 17:16:45 witherspoon netipmid[2864]: << openSession
Aug 03 17:16:45 witherspoon netipmid[2864]: >> RAKP12
Aug 03 17:16:45 witherspoon netipmid[2864]: << RAKP12
Aug 03 17:16:45 witherspoon netipmid[2864]: >> RAKP34
Aug 03 17:16:45 witherspoon netipmid[2864]: << RAKP34
Aug 03 17:16:45 witherspoon netipmid[2864]: >> setSessionPrivilegeLevel
Aug 03 17:16:45 witherspoon netipmid[2864]: << setSessionPrivilegeLevel
Aug 03 17:16:45 witherspoon netipmid[2864]: E> Table:: Command Not found: 0xb03e
```

vishwabmc, 2017-08-22:
@gkeishin Couple suggestions.

- Look into the data that is being sent in that raw dbus command and make sure  0x18 is what you need
- If  0x18  is the correct one, please look into what is in the BootSource. I believe you may have ExternalMedia there. If so, please make it Default and re-test

gkeishin, 2017-08-22:
Sure.. I ll try it on fix it on automation and let you know if it works out or not.. Thanks

gkeishin, 2017-09-18:
We have updated the test code now.. good to close

closed 2017-09-18 by gkeishin