openbmc/openbmc #2194: BMC hangs on r/r when booted to Petitboot

gkeishin, 2017-08-22:
- [x] Boot Host
- [x] RR BMC when reached petitboot prompt
- [x] Observed OOPs and BMC hangs at boot

```
Petitboot (v1.5.1-p641e963)                   8375-22B        WSBMC_008
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
  [Disk: sda2 / 04fe8bcb-5b9c-44ee-ae18-9381aaff9e03]
    Ubuntu, with Linux 4.10.0-20-generic (recovery mode)
    Ubuntu, with Linux 4.10.0-20-generic
    Ubuntu

  System information
  System configuration
  System status log
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell

 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
 Enter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
```

`Serial Console snippet`
```
[ 2414.710000] Unable to handle kernel paging request at virtual address 00560004
[ 2414.710000] pgd = 80004000
[ 2414.710000] [00560004] *pgd=00000000
[ 2414.740000] Internal error: Oops: 805 [#2] ARM
[ 2414.740000] CPU: 0 PID: 926 Comm: systemd-timesyn Tainted: G      D         4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1
[ 2414.740000] Hardware name: ASpeed SoC
[ 2414.740000] task: 9e60b400 task.stack: 9d18c000
[ 2414.740000] PC is at umount_tree+0x60/0x2d4
[ 2414.740000] LR is at drop_collected_mounts+0x3c/0x58
[ 2414.740000] pc : [<8010127c>]    lr : [<80102d44>]    psr: a0000013
[ 2414.740000] sp : 9d18deb8  ip : 9d18def8  fp : 9d18def4
[ 2414.740000] r10: 00000000  r9 : 9c969054  r8 : 00000000
[ 2414.740000] r7 : 9d0dd860  r6 : 9d18dec0  r5 : 00000000  r4 : 00000001
[ 2414.740000] r3 : 9caf6f00  r2 : 08380000  r1 : 9ca7ca28  r0 : 00560000
[ 2414.740000] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2414.740000] Control: 00c5387d  Table: 9ca8c008  DAC: 00000055
[ 2414.740000] Process systemd-timesyn (pid: 926, stack limit = 0x9d18c188)
[ 2414.740000] Stack: (0x9d18deb8 to 0x9d18e000)
[ 2414.740000] dea0:                                                       9d125420 800c1d5c
[ 2414.740000] dec0: 9caf6c48 9d0dd8a8 9d0d0560 806269a8 9d0dd870 ffffe000 9c969020 00000000
[ 2414.740000] dee0: 9c969054 00000000 9d18df0c 9d18def8 80102d44 80101228 9d0ad200 00000000
[ 2414.740000] df00: 9d18df24 9d18df10 80104ac4 80102d14 9cafebe0 00000000 9d18df3c 9d18df28
[ 2414.740000] df20: 80034b94 80104a8c 9e60b400 00000000 9d18df4c 9d18df40 80034ccc 80034b78
[ 2414.740000] df40: 9d18df5c 9d18df50 80034cec 80034c8c 9d18df7c 9d18df60 8001b2f4 80034cdc
[ 2414.740000] df60: 80031e0c 803fee30 9d18c000 000000f8 9d18df94 9d18df80 8001b9fc 8001b084
[ 2414.740000] df80: 00000001 00000000 9d18dfa4 9d18df98 8001ba84 8001b9bc 00000000 9d18dfa8
[ 2414.740000] dfa0: 8000a540 8001ba70 00000001 00000000 00000000 76c650a4 00000000 ffffffff
[ 2414.740000] dfc0: 00000001 00000000 76c62798 000000f8 76c650a4 00000000 54b03e78 00000000
[ 2414.740000] dfe0: 00000000 7ef72ca8 76b55f78 76bc47c4 60000010 00000000 00000000 00000000
[ 2414.740000] [<8010127c>] (umount_tree) from [<80102d44>] (drop_collected_mounts+0x3c/0x58)
[ 2414.740000] [<80102d44>] (drop_collected_mounts) from [<80104ac4>] (put_mnt_ns+0x44/0x64)
[ 2414.740000] [<80104ac4>] (put_mnt_ns) from [<80034b94>] (free_nsproxy+0x28/0x74)
[ 2414.740000] [<80034b94>] (free_nsproxy) from [<80034ccc>] (switch_task_namespaces+0x4c/0x50)
[ 2414.740000] [<80034ccc>] (switch_task_namespaces) from [<80034cec>] (exit_task_namespaces+0x1c/0x20)
[ 2414.740000] [<80034cec>] (exit_task_namespaces) from [<8001b2f4>] (do_exit+0x27c/0x8e8)
[ 2414.740000] [<8001b2f4>] (do_exit) from [<8001b9fc>] (do_group_exit+0x4c/0xb4)
[ 2414.740000] [<8001b9fc>] (do_group_exit) from [<8001ba84>] (SyS_exit_group+0x20/0x24)
[ 2414.740000] [<8001ba84>] (SyS_exit_group) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 2414.740000] Code: e593104c e5930048 e3822302 e5832018 (e5801004)
[ 2414.990000] ---[ end trace 5abe89f6025f9811 ]---
[ 2414.990000] Fixing recursive fault but reboot is needed!
```
- [x] Serial console log 
[ResetBMC_BMC_OS_petitboot.txt](https://github.com/openbmc/openbmc/files/1242344/ResetBMC_BMC_OS_petitboot.txt)

eddiejames, 2017-08-23:
Same issue as #2195

gkeishin, 2017-08-24:
In this case, the BMC didnt recover and hung and I have to AC cycle it, so it could be the same but this behaviour difference is not really promissing..  I can close up but let me know..

rfrandse, 2017-08-30:
fixed Done

closed 2017-08-30 by rfrandse