openbmc/openbmc #2195: Kernel panic - not syncing: Fatal exception in interrupt

gkeishin, 2017-08-22:
- [x] Boot to OS
- [x] Once OS is booted , r/r BMC
- [x] BMC came up, OS is up

but observed this kernel Panic

```
Ubuntu 16.04.2 LTS wsbmc008p1 hvc0

wsbmc008p1 login: root
Password:
Last login: Tue Aug 22 08:45:37 CDT 2017 on hvc0
Welcome to Ubuntu 16.04.2 LTS (GNU/Linux 4.10.0-20-generic ppc64le)

122 packages can be updated.
55 updates are security updates.

root@wsbmc008p1:~#
```

`Serial console snippet`
```
[  408.460000] ------------[ cut here ]------------
[  408.460000] kernel BUG at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/core/skbuff.c:1433!
[  408.460000] Internal error: Oops - BUG: 0 [#2] ARM
[  408.460000] CPU: 0 PID: 1751 Comm: systemd-cgroups Tainted: G      D         4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1
[  408.460000] Hardware name: ASpeed SoC
[  408.460000] task: 9e6004a0 task.stack: 9d1ec000
[  408.460000] PC is at skb_put+0x4c/0x60
[  408.460000] LR is at ftgmac100_poll+0x324/0x4a8
[  408.460000] pc : [<8031c594>]    lr : [<80293678>]    psr: 20000113
[  408.460000] sp : 9d1edca0  ip : 46000000  fp : 9d1edcbc
[  408.460000] r10: 9e50b000  r9 : 00000040  r8 : 00000000
[  408.460000] r7 : 9e50b4d8  r6 : a0c2f280  r5 : 9d786b22  r4 : 00000028
[  408.460000] r3 : 00000000  r2 : 20b00000  r1 : 000000d3  r0 : 9d1366e0
[  408.460000] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  408.460000] Control: 00c5387d  Table: 9d7b0008  DAC: 00000055
[  408.460000] Process systemd-cgroups (pid: 1751, stack limit = 0x9d1ec188)
[  408.460000] Stack: (0x9d1edca0 to 0x9d1ee000)
[  408.460000] dca0: 00000028 9d1366e0 a0c2f280 9e50b4d8 9d1edd04 9d1edcc0 80293678 8031c554
[  408.460000] dcc0: 80614e20 00000000 000000d3 9d1366e0 9e50b480 20b00000 9d1edcf4 9e50b4d8
[  408.460000] dce0: 00000001 00000000 00000040 9e402600 0000000c 0000012c 9d1edd4c 9d1edd08
[  408.460000] dd00: 8032ea10 80293360 00000001 00002a60 9d1edd10 9d1edd10 9d1edd18 9d1edd18
[  408.460000] dd20: 9d1edd54 80614e2c 00000003 00000008 00000100 9e402600 0000000c 9d1ec000
[  408.460000] dd40: 9d1edd9c 9d1edd50 80009554 8032e7cc ffffffff 00000000 9e511d80 00400000
[  408.460000] dd60: 00002a5f 0000000a 80614e20 00000004 9e511d80 00000000 805edb8c 00000000
[  408.460000] dd80: 00000001 9e402600 9d1ec000 00000000 9d1eddac 9d1edda0 8001c5b4 8000949c
[  408.460000] dda0: 9d1eddd4 9d1eddb0 80049e08 8001c4a4 9e403240 00000000 00000001 9d1ede00
[  408.460000] ddc0: 00000020 9d1ec000 9d1eddfc 9d1eddd8 80009488 80049db4 800cc6dc 60000113
[  408.460000] dde0: ffffffff 9d1ede34 800a63f8 9d1ec000 9d1edeb4 9d1ede00 8000e478 80009420
[  408.460000] de00: 00000000 76a5b000 0000000f 9d1ec000 9d0b4c60 00000100 0000000f 76a4c000
[  408.460000] de20: 800a63f8 9d1de0d4 00000000 9d1edeb4 9d1eddf0 9d1ede50 800a6700 800cc6dc
[  408.460000] de40: 60000113 ffffffff 800cca58 7f000000 800cf1e8 800cfa6c 9d1ede8c 9d0b4c60
[  408.460000] de60: 00000054 014200ca 00000000 76a4c000 9d7b1da8 00000000 00000000 00000000
[  408.460000] de80: 00000000 00000000 9d1de0d4 00000000 800d0cbc 9d1edfb0 9e6004a0 9d1de0a0
[  408.460000] dea0: 00000017 76a4c118 9d1edefc 9d1edeb8 80011d30 800cc384 9d1eb640 800e0f14
[  408.460000] dec0: 00001000 9d1eb640 00000000 00000054 76a78000 805d5034 00000017 80011b94
[  408.460000] dee0: 76a4c118 9d1edfb0 7e8606f4 0002cf90 9d1edfac 9d1edf00 80009244 80011ba0
[  408.460000] df00: 00000000 800d3740 9d1edf64 9d1de0d4 9d1eb640 76a78000 00001000 00000003
[  408.460000] df20: 9d1ec000 800c1c84 9d1edf74 9d1edf38 800c1c84 80041f54 00000000 9d1eb640
[  408.460000] df40: 9d1e13c0 800e0f1c 00000000 9d1eb640 9d1edf7c 9d1edf60 800e0f1c 800e48e4
[  408.460000] df60: 9d1e13c0 9d1eb640 6ffffdff 00000006 9d1edf94 9d1edf80 800ff89c 800e0eb8
[  408.460000] df80: 76f07a80 76f09050 9d1edfa4 76ee4758 20000010 ffffffff 00c5387d 00c5387d
[  408.460000] dfa0: 00000000 9d1edfb0 8000e760 80009208 76f07a80 76a4c118 00000006 76a78d3c
[  408.460000] dfc0: 76f07a80 76f09050 6ffffdff 6ffffeff 76f07bd8 7e8606f4 0002cf90 7e860674
[  408.460000] dfe0: 76a4c118 7e860560 76edf2e0 76ee4758 20000010 ffffffff e22ef1db 6b45edfe
[  408.460000] [<8031c594>] (skb_put) from [<80293678>] (ftgmac100_poll+0x324/0x4a8)
[  408.460000] [<80293678>] (ftgmac100_poll) from [<8032ea10>] (net_rx_action+0x250/0x390)
[  408.460000] [<8032ea10>] (net_rx_action) from [<80009554>] (__do_softirq+0xc4/0x2b4)
[  408.460000] [<80009554>] (__do_softirq) from [<8001c5b4>] (irq_exit+0x11c/0x13c)
[  408.460000] [<8001c5b4>] (irq_exit) from [<80049e08>] (__handle_domain_irq+0x60/0xb8)
[  408.460000] [<80049e08>] (__handle_domain_irq) from [<80009488>] (avic_handle_irq+0x74/0x7c)
[  408.460000] [<80009488>] (avic_handle_irq) from [<8000e478>] (__irq_svc+0x58/0x74)
[  408.460000] Exception stack(0x9d1ede00 to 0x9d1ede48)
[  408.460000] de00: 00000000 76a5b000 0000000f 9d1ec000 9d0b4c60 00000100 0000000f 76a4c000
[  408.460000] de20: 800a63f8 9d1de0d4 00000000 9d1edeb4 9d1eddf0 9d1ede50 800a6700 800cc6dc
[  408.460000] de40: 60000113 ffffffff
[  408.460000] [<8000e478>] (__irq_svc) from [<800cc6dc>] (handle_mm_fault+0x364/0xe04)
[  408.460000] [<800cc6dc>] (handle_mm_fault) from [<80011d30>] (do_page_fault+0x19c/0x2c8)
[  408.460000] [<80011d30>] (do_page_fault) from [<80009244>] (do_DataAbort+0x48/0xc8)
[  408.460000] [<80009244>] (do_DataAbort) from [<8000e760>] (__dabt_usr+0x40/0x60)
[  408.460000] Exception stack(0x9d1edfb0 to 0x9d1edff8)
[  408.460000] dfa0:                                     76f07a80 76a4c118 00000006 76a78d3c
[  408.460000] dfc0: 76f07a80 76f09050 6ffffdff 6ffffeff 76f07bd8 7e8606f4 0002cf90 7e860674
[  408.460000] dfe0: 76a4c118 7e860560 76edf2e0 76ee4758 20000010 ffffffff
[  408.460000] Code: e580c090 8a000002 e1a00005 e89da8f0 (e7f001f2)
[  408.460000] ---[ end trace 1475a05af5144e21 ]---
[  408.460000] Kernel panic - not syncing: Fatal exception in interrupt
```
- [x]  Serial console log 
[kernel_panic_rr_OS_booted.txt](https://github.com/openbmc/openbmc/files/1242388/kernel_panic_rr_OS_booted.txt)

eddiejames, 2017-08-23:
# **STATUS**

08/23 - Merged into openbmc linux

rfrandse, 2017-08-30:
Done

closed 2017-08-30 by rfrandse