openbmc/openbmc #2196: yocto build warnings and errors

mdmillerii, 2017-08-22:
After pulling 92b1559c6c69abcb4a60676959f484af8a376d34 and nuking my yocto build directory, I encountered errors similar to the following for 3-4 attempts trying to build witherspoon:
```
DEBUG: Executing python function do_merge_sensor_config
DEBUG: Merging extra configurations: openpower-config.yaml
Traceback (most recent call last):
  File "/tmp/miltonm/build/tmp/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/merge_sensor_config.py", line 12, in <module>
    import yaml
ImportError: No module named yaml
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_merge_sensor_config(d)
     0003:
File: '/esw/san5/miltonm/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-sensor-inventory-mrw-native.bb', lineno: 51, function: do_merge_sensor_config
     0047:        filename = os.path.join(sensoryamldir, url)
     0048:        cmd.append(filename)
     0049:
     0050:    # Invoke the script and don't catch any resulting exception.
 *** 0051:    subprocess.check_call(cmd)
     0052:}
     0053:# python-pyyaml-native is installed by do_configure, so put this task after
     0054:addtask merge_sensor_config after do_configure before do_compile
File: '/esw/san5/iawillia/python/python3/lib/python3.6/subprocess.py', lineno: 291, function: check_call
     0287:    if retcode:
     0288:        cmd = kwargs.get("args")
     0289:        if cmd is None:
     0290:            cmd = popenargs[0]
 *** 0291:        raise CalledProcessError(retcode, cmd)
     0292:    return 0
     0293:
     0294:
     0295:def check_output(*popenargs, timeout=None, **kwargs):
Exception: subprocess.CalledProcessError: Command '['/tmp/miltonm/build/tmp/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/merge_sensor_config.py', '/tmp/miltonm/build/tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor-yamls/config.yaml', '/tmp/miltonm/build/tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor-yamls/openpower-config.yaml']' returned non-zero exit status 1.

Subprocess output:
ERROR: Exception handler error: 'NoneType' object has no attribute 'decode'
DEBUG: Python function do_merge_sensor_config finished
ERROR: Function failed: do_merge_sensor_config
```

After those stopped, I still encountered the following warnings:
```
WARNING: phosphor-logging-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: phosphor-settings-manager-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-phosphor-settings-defaults.populate_sysroot not found
WARNING: openpower-fru-vpd-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: phosphor-led-manager-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: openpower-occ-control-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: phosphor-ipmi-host-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-packagegroup-obmc-ipmid-providers.populate_sysroot not found
WARNING: phosphor-ipmi-host-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-packagegroup-obmc-ipmi-sensors.populate_sysroot not found
WARNING: phosphor-ipmi-host-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: phosphor-ipmi-host-1.0-r1 do_install: Task symlink_create_postinstall changed cwd to /tmp/miltonm/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/image/usr/lib/net-ipmid
WARNING: openpower-ipmi-oem-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-packagegroup-obmc-ipmid-providers.populate_sysroot not found
WARNING: phosphor-ipmi-fru-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-packagegroup-obmc-ipmid-providers.populate_sysroot not found
WARNING: phosphor-ipmi-fru-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-yaml-tiny.populate_sysroot not found
WARNING: phosphor-ipmi-net-1.0-r1 do_configure: /tmp/miltonm/build/tmp/sstate-control/manifest-witherspoon-packagegroup-obmc-ipmid-providers.populate_sysroot not found
WARNING: openpower-ipmi-oem-1.0-r1 do_install: Task symlink_create_postinstall changed cwd to /tmp/miltonm/build/tmp/work/armv6-openbmc-linux-gnueabi/openpower-ipmi-oem/1.0-r1/image/usr/lib/host-ipmid
WARNING: phosphor-ipmi-fru-1.0-r1 do_install: Task symlink_create_postinstall changed cwd to /tmp/miltonm/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/image/usr/lib/host-ipmid

```

mdmillerii, 2017-09-02:
@dhruvibm is the error above related to 1235276ba9b6ecde21437253ec018cc2792625b4 ?

geissonator, 2018-02-09:
I still intermittently hit this error as well.  Solution seems to just keep rebuilding until it fixes itself.

```
ERROR: phosphor-ipmi-sensor-inventory-mrw-native-1.0-r1 do_merge_sensor_config: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_merge_sensor_config(d)
     0003:
File: '/esw/san5/andrewg/Code/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-sensor-inventory-mrw-native.bb', lineno: 52, function: do_merge_sensor_config
     0048:        filename = os.path.join(sensoryamldir, url)
     0049:        cmd.append(filename)
     0050:
     0051:    # Invoke the script and don't catch any resulting exception.
 *** 0052:    subprocess.check_call(cmd)
     0053:}
     0054:# python-pyyaml-native is installed by do_configure, so put this task after
     0055:addtask merge_sensor_config after do_configure before do_compile
File: '/esw/san5/openbmc/python/python3/lib/python3.6/subprocess.py', lineno: 291, function: check_call
     0287:    if retcode:
     0288:        cmd = kwargs.get("args")
     0289:        if cmd is None:
     0290:            cmd = popenargs[0]
 *** 0291:        raise CalledProcessError(retcode, cmd)
     0292:    return 0
     0293:
     0294:
     0295:def check_output(*popenargs, timeout=None, **kwargs):
Exception: subprocess.CalledProcessError: Command '['/tmp/andrewg/02092018-obmc-build/build/tmp/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/merge_sensor_config.py', '/tmp/andrewg/02092018-obmc-build/build/tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor-yamls/config.yaml', '/tmp/andrewg/02092018-obmc-build/build/tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor-yamls/openpower-config.yaml']' returned non-zero exit status 1.

ERROR: phosphor-ipmi-sensor-inventory-mrw-native-1.0-r1 do_merge_sensor_config: Function failed: do_merge_sensor_config
ERROR: Logfile of failure stored in: /tmp/andrewg/02092018-obmc-build/build/tmp/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/temp/log.do_merge_sensor_config.28754
ERROR: Task (/esw/san5/andrewg/Code/openbmc/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-sensor-inventory-mrw-native.bb:do_merge_sensor_config) failed with exit code '1'
Waiting for 1 running tasks to finish:

```

geissonator, 2018-03-23:
Build team hit this one today doing a fresh build

dkodihal, 2018-04-09:
I think https://github.com/openbmc/openbmc/commit/7ca469fd1a491045667efee852838c3a061373d0 should fix this. Can someone confirm if we still run into the issue with this fix?

gtmills, 2018-05-31:
@mdmillerii I assume no one has seen this recently? Call it fixed then. Can this be closed?

mdmillerii, 2018-05-31:
It was supposed to be fixed in April, and I agree that related fixes were merged.  I have not been building to say if it still happens, but will likely need a new list anyways.

closed 2018-05-31 by mdmillerii