openbmc/openbmc #2197: SOL session getting closed automatically after powering on/off

rahulmah, 2017-08-23:
After powering on witherspoon system, SOL session getting closed automatically after some time. 
Same problem is also seen after powering off.

Tested with latest BMC build - 985/v1.99.9-15-g6a7a7a0: 

SOL activate data on powering on.
```
bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H 9.3.181.64 sol activate
[SOL Session operational.  Use ~? for help]

--== Welcome to Hostboot hostboot-be278ac/hbicore.bin ==--

  3.89617|secure|Booting in non-secure mode.
  3.89618|secure|SecureROM valid - enabling functionality
  7.21902|ISTEP  6. 5 - host_init_fsi
  7.33226|ISTEP  6. 6 - host_set_ipl_parms
  7.35543|ISTEP  6. 7 - host_discover_targets
 12.45317|HWAS|PRESENT> DIMM[03]=FFFF000000000000
 12.45317|HWAS|PRESENT> Proc[05]=C000000000000000
 12.49127|ISTEP  6. 8 - host_update_master_tpm
 12.49923|SECURE|Security Access Bit> 0x0000000000000000
 12.49923|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
 12.49934|ISTEP  6. 9 - host_gard
 12.53415|ISTEP  6.10 - host_revert_sbe_mcs_setup
 12.54017|ISTEP  6.11 - host_start_occ_xstop_handler
 12.54110|ISTEP  6.12 - host_voltage_config
 12.57361|ISTEP  7. 2 - mss_volt
 13.09355|ISTEP  7. 3 - mss_freq
 13.22103|ISTEP  7. 4 - mss_eff_config
 16.14057|ISTEP  7. 5 - mss_attr_update
 16.14864|ISTEP  8. 1 - host_slave_sbe_config
 16.21654|ISTEP  8. 2 - host_setup_sbe
 16.22170|ISTEP  8. 3 - host_cbs_start
 16.24334|ISTEP  8. 4 - proc_check_slave_sbe_seeprom_complete
 22.85950|ISTEP  8. 5 - host_attnlisten_proc
 22.86100|ISTEP  8. 6 - host_p9_fbc_eff_config
 22.86615|ISTEP  8. 7 - host_p9_eff_config_links
 22.87613|ISTEP  8. 8 - proc_attr_update
 22.87778|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 22.90693|ISTEP  8.10 - proc_xbus_scominit
 24.46451|ISTEP  8.11 - proc_xbus_enable_ridi
 24.46737|ISTEP  9. 1 - fabric_erepair
 24.47794|ISTEP  9. 2 - fabric_io_dccal
 25.14736|ISTEP  9. 3 - fabric_pre_trainadv
 25.15671|ISTEP  9. 4 - fabric_io_run_training
 25.28101|ISTEP  9. 5 - fabric_post_trainadv
 25.28505|ISTEP  9. 6 - proc_smp_link_layer
 25.28876|ISTEP  9. 7 - proc_fab_iovalid
 25.30750|ISTEP  9. 8 - host_fbc_eff_config_aggregate
 25.31316|ISTEP 10. 1 - proc_build_smp
 25.62963|ISTEP 10. 2 - host_slave_sbe_update
 26.52090|ISTEP 10. 3 - host_set_voltages
 26.56576|ISTEP 10. 4 - proc_cen_ref_clk_enable
 26.57010|ISTEP 10. 5 - proc_enable_osclite
 26.57074|ISTEP 10. 6 - proc_chiplet_scominit
 26.62653|ISTEP 10. 7 - proc_abus_scominit
 26.63779|ISTEP 10. 8 - proc_obus_scominit
 26.69752|ISTEP 10. 9 - proc_npu_scominit
 26.72352|ISTEP 10.10 - proc_pcie_scominit
 26.76766|ISTEP 10.11 - proc_scomoverride_chiplets
 26.77029|ISTEP 10.12 - proc_chiplet_enable_ridi
 26.77291|ISTEP 10.13 - host_rng_bist
 26.77504|ISTEP 10.14 - host_update_redundant_tpm
 26.84389|ISTEP 11. 1 - host_prd_hwreconfig
 26.85101|ISTEP 11. 2 - cen_tp_chiplet_init1
 26.85210|ISTEP 11. 3 - cen_pll_initf
 26.85369|ISTEP 11. 4 - cen_pll_setup
 26.86699|ISTEP 11. 5 - cen_tp_chiplet_init2
 26.86813|ISTEP 11. 6 - cen_tp_arrayinit
 26.87000|ISTEP 11. 7 - cen_tp_chiplet_init3
 26.87111|ISTEP 11. 8 - cen_chiplet_init
 26.87300|ISTEP 11. 9 - cen_arrayinit
 26.87511|ISTEP 11.10 - cen_initf
 26.87609|ISTEP 11.11 - cen_do_manual_inits
 26.87806|ISTEP 11.12 - cen_startclocks
 26.87907|ISTEP 11.13 - cen_scominits
 26.88011|ISTEP 12. 1 - mss_getecid
 26.88487|ISTEP 12. 2 - dmi_attr_update
 26.88607|ISTEP 12. 3 - proc_dmi_scominit
 26.88707|ISTEP 12. 4 - cen_dmi_scominit
 26.88815|ISTEP 12. 5 - dmi_erepair
 26.88994|ISTEP 12. 6 - dmi_io_dccal
 26.89107|ISTEP 12. 7 - dmi_pre_trainadv
 26.89290|ISTEP 12. 8 - dmi_io_run_training
 26.89410|ISTEP 12. 9 - dmi_post_trainadv
 26.89595|ISTEP 12.10 - proc_cen_framelock
 26.89711|ISTEP 12.11 - host_startprd_dmi
 26.89815|ISTEP 12.12 - host_attnlisten_memb
 26.90000|ISTEP 12.13 - cen_set_inband_addr
 26.90112|ISTEP 13. 1 - host_disable_memvolt
 27.24018|ISTEP 13. 2 - mem_pll_reset
 27.25446|ISTEP 13. 3 - mem_pll_initf
 27.29849|ISTEP 13. 4 - mem_pll_setup
 27.32780|ISTEP 13. 6 - mem_startclocks
 27.34483|ISTEP 13. 7 - host_enable_memvolt
 27.34606|ISTEP 13. 8 - mss_scominit
 27.66533|ISTEP 13. 9 - mss_ddr_phy_reset
 27.74484|ISTEP 13.10 - mss_draminit
 27.90115|ISTEP 13.11 - mss_draminit_training
 28.16005|ISTEP 13.12 - mss_draminit_trainadv
 28.16212|ISTEP 13.13 - mss_draminit_mc
 28.18763|ISTEP 14. 1 - mss_memdiag
 34.07283|ISTEP 14. 2 - mss_thermal_init
 34.09055|ISTEP 14. 3 - proc_pcie_config
 34.10671|ISTEP 14. 4 - mss_power_cleanup
 34.11433|ISTEP 14. 5 - proc_setup_bars
 34.15628|ISTEP 14. 6 - proc_htm_setup
 34.16541|ISTEP 14. 7 - proc_exit_cache_contained
 34.18072|ISTEP 15. 1 - host_build_stop_image
 35.17692|ISTEP 15. 2 - proc_set_pba_homer_bar
 35.18089|ISTEP 15. 3 - host_establish_ex_chiplet
 35.18776|ISTEP 15. 4 - host_start_stop_engine
 35.29101|ISTEP 16. 1 - host_activate_master
  0.00715|ISTEP 16. 2 - host_activate_slave_cores
 36.81499|ISTEP 16. 3 - host_secure_rng
 36.84151|ISTEP 16. 4 - mss_scrub
 36.84484|ISTEP 16. 5 - host_ipl_complete
 36.84367|ISTEP 16. 6 - host_load_io_ppe
 36.90245|ISTEP 18.11 - proc_tod_setup
 36.93619|ISTEP 18.12 - proc_tod_init
 36.96696|ISTEP 20. 1 - host_load_payload
 37.19538|ISTEP 20. 2 - host_load_hdat
 40.96293|ISTEP 21. 1 - host_runtime_setup
 94.20551|htmgt|OCCs are now running in ACTIVE state
 94.19051|ISTEP 21. 2 - host_verify_hdat
 94.20729|ISTEP 21. 3 - host_start_payload
SOL session closed by BMC
bash-4.1$
```
If we try to connect SOL again, we get message as SOL is already active.
```
bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H 9.3.181.64 sol activate
Info: SOL payload already active on another session
```

Fw info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-15"
VERSION_ID="v1.99.9-15-g6a7a7a0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-15"
BUILD_ID="v1.99.9"
```

Journal log:
[sol_session_closed_journal.txt](https://github.com/openbmc/openbmc/files/1244245/sol_session_closed_journal.txt)

tomjoseph83, 2017-08-23:
@rahulmah Changes done for #2101 is not available in the build mentioned. Please test it on the latest master.

rahulmah, 2017-08-24:
@tomjoseph83 : I have tested with the latest available master only. Please point to the master build with which I need to retest.

rahulmah, 2017-08-24:
@tomjoseph83 : Also seeing the problem with latest master build 987/v1.99.9-39-g576714d which has fix for #2101

tomjoseph83, 2017-08-30:
@rahulmah Please use the option "usesolkeepalive -N 30" going forward, which is the preferred option when you use ipmi SOL.

sivassrr, 2017-09-18:
@rahulmah do we still see this issue. Or shall we close it?

rahulmah, 2017-10-13:
@sivassrr @tomjoseph83 : SOL works fine with "usesolkeepalive -N 30". But SOL should work with just "sol activate" also. I am keeping this issue open as we need fox for it.

AlexanderAmelkin, 2018-05-31:
Just my 2 cents: on VESNIN hardware, which I believe is based on P8 reference, at least one of the reasons for such a behavior is that with host power going on the Broadcom NC-SI PHY gets reset and also may change speed from 100Mbit to 1GbE due to getting out of power saving mode.

rahulmah, 2019-07-24:
We have open up issue with ipmitool to get it this fixed - https://github.com/ipmitool/ipmitool/issues/146

closed 2020-02-23 by stale[bot]