openbmc/openbmc #2198: [Witherspoon] Rear and front power LED not in ON state at Host booted state

rahulmah, 2017-08-23:
With Host in booted state, rear and front power LED is not coming in "ON" state. It is remains at "Blink" state.

```
bash-4.1$curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/state/enumerate                                                                     {
  "data": {
    "/xyz/openbmc_project/state/bmc0": {
      "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
      "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
      "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
      "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
      "AttemptsLeft": 3,
      "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
      "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
      "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
      "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

```
bash-4.1$ curl -c cjar -b cjar -khttps://<BMC_IP>/xyz/openbmc_project/led/physical/rear_power                                                             {
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Blink"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/front_power
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Blink"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

FW info
```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/cf9fc9cf": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.9-15-g6a7a7a0"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
  },
  "message": "200 OK",
  "status": "ok"
```

vishwabmc, 2017-08-23:
@rahulmah I wonder if you waited enough before checking ? I suggest you wait till Host is Booted prior to checking.

Also, is it an issue on this new driver ?

vishwabmc, 2017-08-23:
I did a fresh build of openbmc and saw that the Priority: field for PowerOn was [Blink] and hence its in that state.

however, the latest version in the `phosphor-mrw-tools` is having the priority addressed.

@spinler I see `https://github.com/open-power/witherspoon-xml/pull/59/commits/47ca614f9b379ce0deead19f0a0fdb270b77f629` is merged. May be its not version bumped yet ?

- This is from build for front-power and rear-power
```
    <attribute>
        <id>LED_PRIORITY</id>
        <default>BLINK</default>
    </attribute>
```
- This is from open-power/witherspoon-xml for front-power and rear-power
```
    <attribute>
        <id>LED_PRIORITY</id>
        <default>ON</default>
    </attribute>

```

That said, it was already [On] in some version of XML.. may be something got messed ?

vishwabmc, 2017-08-23:
# **STATUS**

  - 08/23
     - Analysed and it needs some change in phosphor-mrw-tools. Notified Matt.

spinler, 2017-08-23:
Yea, witherspoon.conf just needs the revision bump.  Will fix.

spinler, 2017-08-23:
https://gerrit.openbmc-project.xyz/#/c/6212/

spinler, 2017-08-23:
This was merged. Closing.

closed 2017-08-23 by spinler