openbmc/openbmc #2200: BMC Update: Write kernel to both chips and update boot env only on success

anoo1, 2017-08-23:


anoo1, 2017-08-25:
- [ ] When creating a kernel volume, create a copy on the alt bmc chip (if an alt chip exists).
- [ ] When deleting a kernel volume, delete the copy from the alt bmc chip (if it exists).
- [ ] Only update the boot env variables with the information of the new volumes if updating the kernel and rootfs finished successfully, this can be done by adding a postExec in the update service file that updates the env variables.
- [ ] Verify that the volumes exist before setting the env variables, this will prevent bricking the system if a volume has been deleted/renamed.

rfrandse, 2017-09-06:
<https://gerrit.openbmc-project.xyz/6426> obmc-flash-bmc: Create a duplicate kernel volume in alt
Resolves  #2200 BMC Update: Write kernel to both chips and update boot env only on success

closed 2017-09-14 by williamspatrick