openbmc/openbmc #2202: OperatingSystemState is inactive on BMC r/r when OS is booted

gkeishin, 2017-08-24:
- [x] Boot to OS
- [x] r/r BMC
- [x] BMC comes up and OS is online

but the `OperatingSystemState ` is `Inactive`
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
        "AttemptsLeft": 3,
        "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified",
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
        "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive",
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
    }
}
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-39"
VERSION_ID="v1.99.9-39-g576714d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-39"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Thu Aug 24 03:21:19 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```
- [x] Journald 
[Os_boot_state_inactive.txt](https://github.com/openbmc/openbmc/files/1247979/Os_boot_state_inactive.txt)

gkeishin, 2017-08-24:
@dkodihal ^^^

dkodihal, 2017-08-24:
I think this has to be addressed with the sensor persistency that @dhruvibm is working on.

rfrandse, 2017-10-12:
@dhruvibm 
@dkodihal said this is this fixed. Can you give me a commit reference?
I'd like to track down the build tag so I can get the CQ into verify state.

rfrandse, 2017-10-12:
verified in ibm-v1.99.10-113-g65edf7d-r1

closed 2017-10-12 by rfrandse